espressif/esp-matter #1209: How to properly shutdown and restart Matter?

bilalmalik76, 2024-12-15:
I am calling this API to shutdown Matter but it crashes
 
chip::Server::GetInstance().Shutdown();

Matter 1.3 and esp-idf 5.2.2

here is a logs: 

`I (33034) chip[SC]: SecureSession[0x3fca8150, LSID:15326]: State change 'kActive' --> 'kPendingEviction'
I (33034) chip[DMG]: AccessControl: finishing
I (33044) chip[DMG]: Examples::AccessControlDelegate::Finish
I (33044) StarPower Main: Commissioning window closed
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

Core  0 register dump:
PC      : 0x4037ea4c  PS      : 0x00060c34  A0      : 0x80381861  A1      : 0x3fcceca0  
--- 0x4037ea4c: esp_cpu_compare_and_set at /home/bilal/Desktop/esp-idf/components/esp_hw_support/cpu.c:459

A2      : 0x3c1d4a44  A3      : 0xffffffff  A4      : 0x63e95c58  A5      : 0x00060c23  
A6      : 0xb33fffff  A7      : 0x0000cdcd  A8      : 0x80382034  A9      : 0x3fccec80  
A10     : 0x3c1d4a44  A11     : 0xb33fffff  A12     : 0x0000cdcd  A13     : 0x00000000  
A14     : 0x3fcce18c  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff  
--- 0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x4037ea49:0x3fcceca0 0x4038185e:0x3fccecd0 0x420b1406:0x3fcced10 0x420b21ed:0x3fcced30 0x420a284e:0x3fcced50 0x420a2c76:0x3fcced80 0x420a314e:0x3fcceda0 0x420a319b:0x3fccedc0 0x420a9b42:0x3fccede0 0x420a255c:0x3fccee00 0x40381eb2:0x3fccee30
--- 0x4037ea49: esp_flash_chip_driver_initialized at /home/bilal/Desktop/esp-idf/components/spi_flash/esp_flash_api.c:253
0x4038185e: vPortEnterCritical at /home/bilal/Desktop/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:571
 (inlined by) xQueueGenericSend at /home/bilal/Desktop/esp-idf/components/freertos/FreeRTOS-Kernel/queue.c:950
0x420b1406: sys_sem_signal at /home/bilal/Desktop/esp-idf/components/lwip/port/freertos/sys_arch.c:134
0x420b21ed: lwip_netconn_do_dns_found at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/api/api_msg.c:2215
0x420a284e: dns_call_found at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/core/dns.c:1026
0x420a2c76: dns_check_entry at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/core/dns.c:1157
0x420a314e: dns_check_entries at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/core/dns.c:1201
 (inlined by) dns_tmr at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/core/dns.c:446
0x420a319b: dns_timeout_cb at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/core/dns.c:430
0x420a9b42: sys_check_timeouts at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/core/timeouts.c:403
0x420a255c: tcpip_timeouts_mbox_fetch at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/api/tcpip.c:98
 (inlined by) tcpip_thread at /home/bilal/Desktop/esp-idf/components/lwip/lwip/src/api/tcpip.c:142
0x40381eb2: vPortTaskWrapper at /home/bilal/Desktop/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

Core  1 register dump:
PC      : 0x4037eb8e  PS      : 0x00060134  A0      : 0x820045fc  A1      : 0x3fcc0ae0  
--- 0x4037eb8e: esp_cpu_wait_for_intr at /home/bilal/Desktop/esp-idf/components/esp_hw_support/cpu.c:145

A2      : 0x00000000  A3      : 0x3fcc0cb4  A4      : 0x80382174  A5      : 0x3fce1ce0  
A6      : 0x00060023  A7      : 0x00000003  A8      : 0x820ed846  A9      : 0x3fcc0aa0  
A10     : 0x00000000  A11     : 0x3fcc0cb4  A12     : 0x80382174  A13     : 0x3fce1790  
A14     : 0x00060023  A15     : 0x00000003  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4037eb8b:0x3fcc0ae0 0x420045f9:0x3fcc0b00 0x40383323:0x3fcc0b20 0x40381eb2:0x3fcc0b40
--- 0x4037eb8b: xt_utils_wait_for_intr at /home/bilal/Desktop/esp-idf/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at /home/bilal/Desktop/esp-idf/components/esp_hw_support/cpu.c:132
0x420045f9: esp_vApplicationIdleHook at /home/bilal/Desktop/esp-idf/components/esp_system/freertos_hooks.c:59
0x40383323: prvIdleTask at /home/bilal/Desktop/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:4307 (discriminator 1)
0x40381eb2: vPortTaskWrapper at /home/bilal/Desktop/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:134

ELF file SHA256: e456642b5

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375c34
--- 0x40375c34: esp_restart_noos at /home/bilal/Desktop/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:159

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x178c
load:0x403c9700,len:0x4
load:0x403c9704,len:0xcbc
load:0x403cc700,len:0x2d9c
entry 0x403c9914`

shubhamdp, 2024-12-18:
@bilalmalik76 We are looking into this, will check and get back.

bilalahmaddev, 2025-01-24:
@shubhamdp any update on this?

shubhamdp, 2025-01-24:
@bilalahmaddev work is going on in https://github.com/project-chip/connectedhomeip/pull/36972, will need some time as we are fixing multiple issues that are coming across.

bilalahmaddev, 2025-04-29:
@shubhamdp any update on this as this is important for us.