espressif/esp-matter #1210: Can I send the temperature sensor data to AWS IOT cloud using ESP32H2 and Homepod?

GayathriHommi, 2024-12-16:
I am using ESP32h2, a temperature sensor, homepod and I'm using esp-matter code example- sensors for sending the temperature data to Homepod home application. I want to send this temperature data to AWS iot cloud through VScode can i get any reference code or documents

wqx6, 2024-12-18:
Hi, @GayathriHommi you can connect the ESP32-H2 to the cloud server with Homepod as the Apple Homepod supports the NAT64 feature. To use this feature your need to enable the following options in menuconfig:
```
CONFIG_OPENTHREAD_DNS64_CLIENT=y
CONFIG_LWIP_HOOK_DNS_EXT_RESOLVE_CUSTOM=y
```
After that you can use the common APIs to access Internet.

pimpalemahesh, 2025-01-10:
@GayathriHommi  Please close the issue if it has been resolved.

GayathriHommi, 2025-01-16:
@wqx6  while trying to connect esp32h2 to aws iot i'm facing some issues but the same code is working fine with esp32s3 
 
esp-tls: couldn't get hostname for :a1q5vtttjz7alq-ats.iot.ap-south-1.amazonaws.com: getaddrinfo() returns 202, addrinfo=0x0
E (90334) esp-tls: Failed to open new connection
E (90334) transport_base: Failed to open a new connection
E (90344) mqtt_client: Error transport connect

Are there any more changes to be made other than this 
CONFIG_OPENTHREAD_DNS64_CLIENT=y
CONFIG_LWIP_HOOK_DNS_EXT_RESOLVE_CUSTOM=y

wqx6, 2025-01-22:
> [@wqx6](https://github.com/wqx6) while trying to connect esp32h2 to aws iot i'm facing some issues but the same code is working fine with esp32s3

Are you using Homepod as the Thread BR for your Thread network? Note that Thread Border Router should support NAT64 so that the End device can resolve an URL's IP address. Currently Homepod supports NAT64 but Nesthub doesn't. Did the ESP32-H2 join the thread network of TBR? Could you upload the log file of ESP32-H2?

> Are there any more changes to be made other than this

No, these configurations are enough for a Thread node to connect Internet.

GayathriHommi, 2025-02-09:
hommi/PC:~/esp/Projects/AWS_proj/Work_updates/231124/sensors_maac_s3_room_23012024$ idf.py build flash monitor
Executing action: all (aliases: build)
Running ninja in directory /home/hommi/esp/Projects/AWS_proj/Work_updates/231124/sensors_maac_s3_room_23012024/build
Executing "ninja all"...
[0/1] Re-running CMake...
-- Component directory /home/hommi/esp/esp-matter/examples/common/blemesh_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/hommi/esp/esp-matter/examples/common/cmake_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/hommi/esp/esp-matter/examples/common/external_platform does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/hommi/esp/esp-matter/examples/common/relinker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/hommi/esp/esp-matter/examples/common/secondary_ble_adv does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/hommi/esp/esp-matter/examples/common/utils does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- Building ESP-IDF components for target esp32h2
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
NOTICE: Processing 24 dependencies:
NOTICE: [1/24] espressif/button (3.5.0)
NOTICE: [2/24] espressif/cmake_utilities (0.5.3)
NOTICE: [3/24] espressif/esp-serial-flasher (0.0.11)
NOTICE: [4/24] espressif/esp_bsp_generic (1.2.1)
NOTICE: [5/24] espressif/esp_delta_ota (1.1.0)
NOTICE: [6/24] espressif/esp_encrypted_img (2.1.0)
NOTICE: [7/24] espressif/esp_lcd_gc9a01 (1.2.0)
NOTICE: [8/24] espressif/esp_lcd_ili9341 (1.2.0)
NOTICE: [9/24] espressif/esp_lcd_touch (1.1.2)
NOTICE: [10/24] espressif/esp_lcd_touch_cst816s (1.0.6)
NOTICE: [11/24] espressif/esp_lcd_touch_ft5x06 (1.0.6~1)
NOTICE: [12/24] espressif/esp_lcd_touch_gt1151 (1.0.5~2)
NOTICE: [13/24] espressif/esp_lcd_touch_gt911 (1.1.1~2)
NOTICE: [14/24] espressif/esp_lcd_touch_tt21100 (1.1.0~1)
NOTICE: [15/24] espressif/esp_lvgl_port (2.4.4)
NOTICE: [16/24] espressif/esp_rcp_update (1.2.0)
NOTICE: [17/24] espressif/esp_secure_cert_mgr (2.5.0)
NOTICE: [18/24] espressif/jsmn (1.1.0)
NOTICE: [19/24] espressif/json_generator (1.1.2)
NOTICE: [20/24] espressif/json_parser (1.0.3)
NOTICE: [21/24] espressif/led_indicator (0.9.3)
NOTICE: [22/24] espressif/led_strip (3.0.0)
NOTICE: [23/24] lvgl/lvgl (9.2.2)
NOTICE: [24/24] idf (5.2.1)
-- Found Python3: /home/hommi/.espressif/python_env/idf5.2_py3.12_env/bin/python3.12 (found version "3.12.3") found components: Interpreter 
NOTICE: Skipping optional dependency: espressif/cmake_utilities
NOTICE: Skipping optional dependency: espressif/mdns
NOTICE: Skipping optional dependency: espressif/esp_insights
-- Project sdkconfig file /home/hommi/esp/Projects/AWS_proj/Work_updates/231124/sensors_maac_s3_room_23012024/sdkconfig
warning: default on the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/hommi/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:979, /home/hommi/esp/esp-matter/components/esp_matter/Kconfig:48) will have no effect, as defaults do not affect choice symbols
warning: the choice symbol SEC_CERT_DAC_PROVIDER (defined at /home/hommi/esp/esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/Kconfig:979, /home/hommi/esp/esp-matter/components/esp_matter/Kconfig:48) is defined with a prompt outside the choice
Loading defaults file /home/hommi/esp/Projects/AWS_proj/Work_updates/231124/sensors_maac_s3_room_23012024/sdkconfig.defaults...
Loading defaults file /home/hommi/esp/Projects/AWS_proj/Work_updates/231124/sensors_maac_s3_room_23012024/sdkconfig.defaults.esp32h2...
-- App "sensors" version: 1.0
-- Adding linker script /home/hommi/esp/Projects/AWS_proj/Work_updates/231124/sensors_maac_s3_room_23012024/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_system/ld/esp32h2/sections.ld.in
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.ld
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.api.ld
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.libgcc.ld
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.newlib.ld
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.version.ld
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.wdt.ld
-- Adding linker script /home/hommi/esp/esp-idf/components/esp_rom/esp32h2/ld/esp32h2.rom.heap.ld
[... cut ...]
I (130068) shtc3: RH float= 4294967296.000000
I (130068) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000402's Attribute 0x00000000 is -4735 **********
E (130078) app_main: AWS_temp -47.349998
I (130088) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000405's Attribute 0x00000000 is null **********
I (132068) shtc3: Temperature Raw= 0,0

Done
hommi@hommi-HP-ZBook-Firefly-14-inch-G8-Mobile-Workstation-PC:~/esp/Projects/AWS_proj/Work_updates/231124/sensors_maac_s3_room_23012024$

GayathriHommi, 2025-02-09:
homepod is of latest model MQJ73HNJA, device connected to thread