espressif/esp-matter #1212: Matter Unique factory partition data and how to Manage

bilalmalik76, 2024-12-18:
For production of Matter devices based on ESP32-S3, how to generate unique partition data and manage these in production? Our first Matter device is in process of certification, we want to know how we can handle this.

Thanks

shubhamdp, 2024-12-19:
@bilalmalik76 Please take a look at below blogs from Espressif. You will get an idea about the how to generate/provision the Matter unique Matter data (along with any manufacturer specific data) in the device. Please let me know if you have further queries.

- [Accelerating Matter Device Manufacturing](https://developer.espressif.com/blog/accelerating-matter-device-manufacturing/)
- [Matter Improvements to Espressif DAC Provisioning Service](https://developer.espressif.com/blog/matter-improvements-to-espressif-dac-provisioning-service/)

dhrishi, 2024-12-20:
I am closing this as the question is answered. Please re-open in case of any issues

closed 2024-12-20 by dhrishi