espressif/esp-matter #1214: mDNS server issues when using the Matter Contoller application  with Thread

pavel808, 2024-12-18:
I am trying to customize the Matter Controller example which is running on an ESP Thread Border Router device. 

I can't seem to get an mDNS server working with this. See my `app_main()` and `initialise_mdns(void)` below. 
Everything appears to run fine and mDNS initializes ok, but not possible to ping "my-server" on the local network no matter what.

Before building the application project, I configure and set the target as follows : 
`idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.otbr" set-target esp32s3 `. 

My `sdkconfig.defaults.otbr` is attached also. 

Then I configure the Wi-Fi credentials on the terminal via : `matter esp wifi connect {ssid} {password} `. 
These credentials of course then stay saved on the device unless I do a complete erase and flash. 

I have built and flashed other mDNS examples such as `esp-idf/examples/protocols/esp_local_ctrl` which work fine and can ping the hostname. 
One of the differences I see with those examples is that `example_connect()` is used to get the  Wi-Fi  credentials. If I try to use `example_connect()` with my below code I get a crash. 

I have attached an `app-output.log` file too which shows the terminal output from my application. I'd appreciate any help with this. Thanks
 
```
static void initialise_mdns(void)
{
    char *hostname = "my-server";

    //initialize mDNS
    ESP_ERROR_CHECK( mdns_init() );
    ESP_ERROR_CHECK( mdns_hostname_set(hostname) );
    ESP_LOGI(TAG, "MDNS hostname set to: %s", hostname);
}

extern "C" void app_main()
{
    esp_err_t err = ESP_OK;

    /* Initialize the ESP NVS layer */ 
    nvs_flash_init();

#if CONFIG_ENABLE_CHIP_SHELL
    esp_matter::console::diagnostics_register_commands();
    esp_matter::console::wifi_register_commands();
    esp_matter::console::init();
#if CONFIG_ESP_MATTER_CONTROLLER_ENABLE
    esp_matter::console::controller_register_commands();
#endif // CONFIG_ESP_MATTER_CONTROLLER_ENABLE
#ifdef CONFIG_OPENTHREAD_BORDER_ROUTER
    esp_matter::console::otcli_register_commands();
#endif // CONFIG_OPENTHREAD_BORDER_ROUTER
#endif // CONFIG_ENABLE_CHIP_SHELL
#ifdef CONFIG_OPENTHREAD_BORDER_ROUTER
#ifdef CONFIG_AUTO_UPDATE_RCP
    esp_vfs_spiffs_conf_t rcp_fw_conf = {.base_path = "/rcp_fw", .partition_label = "rcp_fw", .max_files = 10, .format_if_mount_failed = false};
    if (ESP_OK != esp_vfs_spiffs_register(&rcp_fw_conf)) 
    {
        ESP_LOGE(TAG, "Failed to mount rcp firmware storage");
	return;
    }
    esp_rcp_update_config_t rcp_update_config = ESP_OPENTHREAD_RCP_UPDATE_CONFIG();
    openthread_init_br_rcp(&rcp_update_config);
#endif
    /* Set OpenThread platform config */
    esp_openthread_platform_config_t config = 
    {
	      .radio_config = ESP_OPENTHREAD_DEFAULT_RADIO_CONFIG(),
	      .host_config = ESP_OPENTHREAD_DEFAULT_HOST_CONFIG(),
	      .port_config = ESP_OPENTHREAD_DEFAULT_PORT_CONFIG(),
    };
    set_openthread_platform_config(&config);
#endif // CONFIG_OPENTHREAD_BORDER_ROUTER
	  /* Matter start */
    err = esp_matter::start(app_event_cb);
    ABORT_APP_ON_FAILURE(err == ESP_OK, ESP_LOGE(TAG, "Failed to start Matter, err:%d", err));

#if CONFIG_ESP_MATTER_COMMISSIONER_ENABLE
    esp_matter::lock::chip_stack_lock(portMAX_DELAY);
    esp_matter::controller::matter_controller_client::get_instance().init(112233, 1, 5580);
    esp_matter::controller::matter_controller_client::get_instance().setup_commissioner();
    esp_matter::lock::chip_stack_unlock();
#endif // CONFIG_ESP_MATTER_COMMISSIONER_ENABLE

  //ESP_ERROR_CHECK(esp_netif_init());
  initialise_mdns();

  while(1)
  {
      sleep(5);
  }

}
```
[app-output.log](https://github.com/user-attachments/files/18178600/app-output.log)

[sdkconfig.defaults.otbr.txt](https://github.com/user-attachments/files/18178598/sdkconfig.defaults.otbr.txt)

wqx6, 2024-12-27:
The matter will set the hostname to the MAC address of the device after the device's IP addresses is changed, that's why you cannot ping `my_server`. And this is required by Matter SPEC, so please don't use the `mdns_hostname_set()` to change the host name.

If you want to add another hostname, you can add a delegated host with [mdns_delegate_hostname_add()](https://github.com/espressif/esp-protocols/blob/master/components/mdns/include/mdns.h#L173).

pavel808, 2024-12-28:
@wqx6 Thanks as always. 

Based on that. I've finally solved it with my below function. Hopefully it may help others also : 

```
static void mdns_delegate_hostname(void)
{
	esp_err_t err = ESP_OK;

	char *delegated_hostname = "my-esp-server";

	esp_netif_t *intf = esp_netif_get_default_netif();;

	while (intf == NULL)
	{
		 ESP_LOGE(TAG, "ERROR ! netif is NULL");
		 intf = esp_netif_get_default_netif();
		 sleep(2);
	}

	mdns_ip_addr_t addr4, addr6;
	addr4.addr.type = ESP_IPADDR_TYPE_V4;
	esp_netif_ip_info_t info;
	esp_netif_get_ip_info(intf, &info);

	ESP_LOGI(TAG, "IP Address of device is : " IPSTR "\n", IP2STR(&info.ip));

	addr4.addr.u_addr.ip4 = info.ip;
	addr6.addr.type = ESP_IPADDR_TYPE_V6;
	esp_netif_get_ip6_linklocal(intf, &addr6.addr.u_addr.ip6);
	addr4.next = &addr6;
	addr6.next = NULL;

	ESP_LOGI(TAG, "Setting delegated hostname to %s\n", delegated_hostname);

	err = mdns_delegate_hostname_add(delegated_hostname, &addr4);
	if (err != ESP_OK)
	{
		ESP_LOGE(TAG, "ERROR Setting delegated hostname to %s\n", delegated_hostname);
		return;
	}

	err = mdns_service_add_for_host(NULL, "_http", "_tcp", delegated_hostname, 80, NULL, 0);
	if (err != ESP_OK)
	{
		ESP_LOGE(TAG, "ERROR adding service for host %s\n", delegated_hostname);
	}
}

```

closed 2024-12-28 by pavel808