espressif/esp-matter #1216: Matter Controller discovery of commissioned devices

pavel808, 2024-12-19:
Is there currently a way for the Matter Controller Client to be able to discover all of the commissioned devices for that controller when requested? The equivalent of using `./chip-tool discover`

The `matter_controller_client` has a `DeviceCommissioner` and `DeviceController` class but I can't find any public APIs within these for this. Only discovery of advertising commissionable devices?

My use case is basically to send my Matter Controller Client application a message via MQTT at any time which should then discover and list the currently commissioned devices for that controller.

wqx6, 2024-12-23:
Normally the controller gets the device's node id from cloud and then establish secure session with the device use that node id.

But Matter provide APIs to browse operational nodes and a console command which uses those APIs. You can try it with the console command `matter dns browse operational _I{compressed_fabric_id}`.  Note that the compressed fabric id is the controller's fabric ID.

dhrishi, 2025-01-10:
Closing as the question is answered. Please re-open for follow-up questions

closed 2025-01-10 by dhrishi