espressif/esp-matter #1218: IOT button uses too much power

AlexVaq, 2024-12-20:
I'm trying to design a low-power, sleepy thread device for homekit, based on the icd_app example of esp-matter.I'm currently working with a devboard esp32c6 devkitM-1.

I noticed that if I initialize the button (which is quite useful to unpair from the thread network, reboot, etc), the minimum consumption I measure is around 3-4 mA, whereas if I don't, the consumption goes back down to the microA level.
I don't think this is intended. Maybe this is a specific problem of the development board, but I'm inclined to think that the code related to the iot button is not as efficient as it should.

Apart from the esp32c6 devkitM-1 board I'm using esp-idf 5.4 and I'm installing esp-matter directly from vscode.

The code I use to initialize the button is the following:

```
app_driver_handle_t AppDriverButtonInit()
{
    /* Initialize button */

    button_config_t config = button_driver_get_config();
    button_handle_t handle = iot_button_create(&config);

    iot_button_register_cb(handle, BUTTON_PRESS_DOWN,        AppDriverButtonToggleCb,           NULL);
    iot_button_register_cb(handle, BUTTON_LONG_PRESS_HOLD,   AppDriverButtonLongPressCb,        NULL);
    iot_button_register_cb(handle, BUTTON_PRESS_UP,          AppDriverButtonLongPressReleaseCb, NULL);
    iot_button_register_cb(handle, BUTTON_DOUBLE_CLICK,      AppDriverButtonDoubleClickCb,      NULL);
    iot_button_register_cb(handle, BUTTON_PRESS_REPEAT,      AppDriverButtonRepeatCb,           NULL);
    iot_button_register_cb(handle, BUTTON_PRESS_REPEAT_DONE, AppDriverButtonRepeatDoneCb,       NULL);

    return (app_driver_handle_t)handle;
}
```

but the consumption is the same even if I just call iot_button_create without registering any callbacks. The only workaround I found is to ignore the button functionality.

wqx6, 2024-12-23:
It seems that the iot_button component prevents the device from entering sleep mode, resulting in a current of approximately 3-4 mA. We will add a button for the ICD example in a later update so that you can use that button for your application. 
Before that you can refer to the button class in the upstream repo https://github.com/project-chip/connectedhomeip/blob/master/examples/lit-icd-app/esp32/main/include/IcdUatButton.h and use it for your device.

AlexVaq, 2024-12-23:
That makes a lot of sense, thank you so much.
I think I should keep the issue opened until the button is added to the ICD example in the esp-matter repo.
I'll keep an eye on it.

wqx6, 2024-12-26:
Hi, could you enable `CONFIG_GPIO_BUTTON_SUPPORT_POWER_SAVE` in menuconfig and add `config.gpio_button_config.enable_power_save = true;` before the code you created the iot_button? In my test the device can enter sleep mode with that configuration. Note that only RTC GPIOs can be used for wake up the device, For ESP32-H2, they are GPIO9-GPIO14, and for ESP32-C6, they are GPIO0-GPIO7.

I will still add the code initializing the iot_button to the icd example.

AlexVaq, 2024-12-26:
Thanks, @wqx6. I didn't find the option CONFIG_GPIO_BUTTON_SUPPORT_POWER_SAVE anywhere in menuconfig. Is it in a new commit? Is it a hidden option I have to manually add in the sdkconfig file?

AlexVaq, 2024-12-26:
Hi, @wqx6.

Not only couldn't I find the menuconfig option, but I didn't find the config.gpio_button_config.enable_power_save field anywhere either.
In the file esp-matter/device_hal/button_driver/button/include//button_gpio.h there is not such a field, no matter which branch of esp-matter you check.

Maybe you forgot to commit the changes?

wqx6, 2024-12-27:
The iot_button component is [this repository](https://github.com/espressif/esp-iot-solution/tree/f5ca5b553873e8ff3c044d5bae56d0eae1d0673b/components/button) and you can find the config option at the [Kconfig file](https://github.com/espressif/esp-iot-solution/blob/f5ca5b553873e8ff3c044d5bae56d0eae1d0673b/components/button/Kconfig#L34). If you don't find the menuconfig option, could you check the button component version? You can use the version v3.4.1 which I have verified being able to work with light sleep mode.

AlexVaq, 2024-12-29:
Thanks for all your efforts, @wqx6. I had to modify the button implementation of esp-matter in device_hal to be based on iot_button 3.4.1 (the implementation in device_hal requires iot_button = 2.5.0), and I managed to make it compile. Unfortunately the devkit boards for esp32c6 use GPIOs 8 and 9 for the BOOT button, so it seems there is no way to do powersaving there, and I get a non-catastrophic error during initialization. It looks like the esp32h2 devkit can benefit from this feature, but not the esp32c6 devkits.

I guess there isn't much I can do, unless I make my own board.

itatboy, 2024-12-30:
@AlexVaq 

I have encountered a similar problem, and the details are:

I pulled the commit: a07c3129aa5f9889eb173ed52e4d5a6a904f14bc
No code changes were made, compile esp-matter/examples/icd_app, and download it to esp32-h2-devkitm-1-v1.2.
The current obtained from the test is around 5mA and has not reached uA.

Can you provide some assistance? Thank

This is the wiring diagram:
[image]

AlexVaq, 2024-12-30:
Because the button component esp-matter uses is still version 2.5.0, and the power saving feature was implemented in a later version (latest version of iot_button is 3.4.1). You can't add the iot_button 3.4.1 as a component, because esp-matter request specifically the 2.5.0 version and there will be a conflict, so you must replace the requirement in device_hal/button_driver/iot_button/idf_component.yml from 2.5.0 to 3.4.1, run cmake again, and then go to menuconfig and enable the CONFIG_GPIO_BUTTON_SUPPORT_POWER_SAVE option. When initializing the button you must make sure to set

`config.gpio_button_config.enable_power_save = true;`

Then it should work, although because my board is not supported, I couldn't check it.

wqx6, 2025-01-02:
@itatboy Could you tell me what peripherals are your using for the devkit board? Note that the light sleep of BLE requires the 32kHz external crystal. If your devkit board doesn't have a 32kHz external crystal, the chip cannot enter sleep mode before commissioning is completed(as BLE is advertising before commissioning).

wqx6, 2025-01-02:
@AlexVaq Hi, The button support has already been added in the icd example, I have tested both ESP32-H2 and ESP32-C6 and both the chip works well with the default button PIN(For ESP32-H2 it is GPIO9 which is the boot button and for ESP32-C6 it is GPIO7).

> Unfortunately the devkit boards for esp32c6 use GPIOs 8 and 9 for the BOOT button, so it seems there is no way to do powersaving there.

Yes, only RTC GPIOs can be used for wake up the device. You should connect an external button to the button GPIO if you are using our ESP32-C6 DebKit boards.

AlexVaq, 2025-01-02:
Thank you, @wqx6. That solves all my problems, but since @itatboy is still asking questions, I won't close the issue yet.

itatboy, 2025-01-03:
@AlexVaq 
Thank you for your help. I will confirm the version of iot-button and switch to testing 3.4.1. Any progress will be updated here

itatboy, 2025-01-03:
@wqx6 
I used the official ESP32-H2-DovKitM-1 from Lexin, with the following peripherals:
[image]

**ESP32-H2-DevKitM-1**
https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32h2/esp32-h2-devkitm-1/index.html

itatboy, 2025-01-03:
@AlexVaq @wqx6 

The SIT waveform captured using ESP32-H2-DevKitM-1 has an average current of 86.26uA. If this waveform is fine, the issue can be close. 
Thanks

[image]

itatboy, 2025-01-05:
@tomasmcguinness 
I guess problem in 

> sdkconfig. Defaults. Esp32h2. Lit 

Add:
```
CONFIG_ESP_WIFI_SOFTAP_SUPPORT=n
CONFIG_ENABLE_WIFI_AP=n
```
It can be successfully compiled, but when I tested, the power consumption of ESP32-H2 continues to be around 800uA, and SIT is not as good as LIT, which is the conclusion I got after testing.

@wqx6 
With SIT, another problem was encountered: ESP32-H2 was woked up for a considerable amount of time, which was clearly not normal.
[image]

wqx6, 2025-01-08:
[image]
@itatboy From the picture that you shared, the device enters active mode for about 2 mins, it could be caused by the above reasons. And it is normal as the device returns Idle mode at last. It is expected that LIT ICD has better power save performance  than SIT ICD, as the Slow polling interval is much longer for LIT ICDs

itatboy, 2025-01-08:
Regarding the power consumption of ESP32-H2, I will continue to test it. Now, I have no problem.

Thank @wqx6 @AlexVaq