espressif/esp-matter #1219: TC-ACE-1.4 failing

bilalmalik76, 2024-12-22:
[image]
[TC-ACE-1.4 failing.log](https://github.com/user-attachments/files/18221636/TC-ACE-1.4.failing.log)

I have passed this in config: 

` "test_parameters": {
      "paa_trust_store_path": "/paa-root-certs/",
      "storage_path": "/root/admin_storage.json",
      "int_arg": "PIXIT.ACE.APPENDPOINT:1 PIXIT.ACE.APPDEVTYPEID:256",
      "string_arg": "PIXIT.ACE.APPCLUSTER:OnOff PIXIT.ACE.APPATTRIBUTE:OnOff"
    }`

VaishaliAvhale, 2024-12-24:
Hi @bilalmalik76 , I went through the failure logs and noticed that the string_arg parameter seems to be missing in the test case execution command. Could you please check the test_parameter configuration?

For executing TC_ACE_1_4, we only need to provide the following two parameters:
[image]

dhrishi, 2025-01-02:
@bilalmalik76 Can you please share the requested details? Or close the issue if resolved

bilalmalik76, 2025-01-02:
@dhrishi I have already set this in config file:

` "test_parameters": { "paa_trust_store_path": "/paa-root-certs/", "storage_path": "/root/admin_storage.json", "int_arg": "PIXIT.ACE.APPENDPOINT:1 PIXIT.ACE.APPDEVTYPEID:256", "string_arg": "PIXIT.ACE.APPCLUSTER:OnOff PIXIT.ACE.APPATTRIBUTE:OnOff" }`

I am using matter 1.3 and v2.10.2+spring2024.

VaishaliAvhale, 2025-01-02:
@bilalmalik76  Please share your Test Harness project configuration file. It seems the `string_arg` is not being passed during test execution.

bilalahmaddev, 2025-01-13:
we can close this.

closed 2025-01-13 by bilalahmaddev