espressif/esp-matter #1220: Generic Switch example only supports single press

hvolpers, 2024-12-24:
**Describe the bug**
I build/installed the generic_switch example to an ESP32-C6 board using a GPIO Button (same behaviour will the boot button which is default). On bootup I get the message:
```
I (659) app_main: Generic Switch created with endpoint_id 1
E (659) esp_matter_feature: Momentary switch multi press is not supported because momentary switch and/or momentary switch releaseis absent
```

After adding the device to HomeKit (all devices updated to latest releases) single press is the only option available.

The error message seams to be iot_button related. I'm new to ESP32/Matter and I wasn't able to figure out what exacly the problem is.

**Environment**
- ESP-Matter Commit Id: 96cb173cb49f0f58bee16f630543cc3c9276ce43
- ESP-IDF Commit Id: v5.3.2 (9d7f2d69f50d1288526d4f1027108e314e8c879f)
- SoC (eg: ESP32 or ESP32-C3): ESP32-H2 (Waveshare ESP32-C6-DEV-KIT-N8)
- Device Logs (Please attach the log file): s. above, let me know if I can provide more information

- Host Machine OS: Ubuntu 24.10
- Host Machine Python version: 3.12.7

Jerry-ESP, 2025-01-02:
Did you use the default generic switch example? Can you share your code of the app_main.cpp ?

Jerry-ESP, 2025-01-02:
The current generic switch example support  MomentarySwitch, MomentarySwitchMultiPress and ActionSwitch features, but the ActionSwitch is updated in matter1.4, maybe Apple home has not supported this feature. In matter1.3, If you want to add the MomentarySwitchMultiPress feature, you must add MomentarySwitchRelease feature.

Jerry-ESP, 2025-01-02:
In matter1.3
[image]

In matter1.4
[image]

hvolpers, 2025-01-02:
Hi,

thanks for trying to help me out. Looks like I messed something up, as I wasn't able to reproduce the problem with a fresh copy of the generic_button example.
However, how can I add the long-press capability? As you pointed out (and that is what I found in the code as well) the combination Momentary Switch & Action Switch should be enough to add the long press feature, right?
I added the long press feature in app_main.cpp:
```
#if CONFIG_GENERIC_SWITCH_TYPE_MOMENTARY                                                                                                                                                                                                                                                              
    cluster::switch_cluster::feature::momentary_switch::add(cluster);                                                                                                                                                                                                                                 
    cluster::switch_cluster::feature::action_switch::add(cluster);                                                                                                                                                                                                                                    
    cluster::switch_cluster::feature::momentary_switch_multi_press::config_t msm;                                                                                                                                                                                                                     
    msm.multi_press_max = 5;                                                                                                                                                                                                                                                                          
    cluster::switch_cluster::feature::momentary_switch_multi_press::add(cluster, &msm);                                                                                                                                                                                                               
    cluster::switch_cluster::feature::momentary_switch_long_press::add(cluster);                                                                                                                                                                                                                      
#endif                                                                                                                                                                                                                                                                                                
```

but after rebuilding and flashing on boot up I get this (error) message from the ESP (erase-flash, flash, not coupled, first start right after flashing)
```
[...]
I (660) app_main: Generic Switch created with endpoint_id 1
E (660) esp_matter_feature: Momentary switch long press is not supported.
I (670) pp: pp rom version: 5b8dcfa
[...]
```

Sorry for being a c/c++/esp32 n00b here (working on it ;)). What am I missing here?

Thanks for taking the time and sorry for me messing something up. I will create a clean test case next time before opening an issue.

Jerry-ESP, 2025-01-02:
Hi 
    This is a known issue, we will fix it soon, you can apply this patch and have a try.

 Thanks!
[esp-matter-fix-switch-cluster-feature.zip](https://github.com/user-attachments/files/18290283/esp-matter-fix-switch-cluster-feature.zip)

hvolpers, 2025-01-02:
Thx, I will give it a try. Something else I realised: After removing the device I am not able to couple it again. The only way I found was to eraseflash and flash again. Any links/pointers for me to learn how to setup the full device lifecycle?

Jerry-ESP, 2025-01-02:
You can factory reset the device by type in the cli: matter device factoryreset, then you can pairing it again.

hvolpers, 2025-01-02:
Sure. But that is not exacly an end-user experience, is it? ;) So I guess what you are saying that I have to implement the reset button myself as it is not in scope for the examples?

closed 2025-01-02 by hvolpers

Jerry-ESP, 2025-01-02:
> Sure. But that is not exacly an end-user experience, is it? ;) So I guess what you are saying that I have to implement the reset button myself as it is not in scope for the examples?

Yes, but you can refer to the light example to implement the long press factoryreset function: https://github.com/espressif/esp-matter/blob/96b09392313da50f5e8bce6cc0a52992b602834f/examples/light/main/app_main.cpp#L156

hvolpers, 2025-01-02:
I'll look into the light example. Thank you for helping me out and providing the fix, very much appreciated.