espressif/esp-matter #1222: switch_cluster::create() missing final parameter for feature map

jonsmirl, 2024-12-26:
https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_cluster.cpp#L1917
~~~
namespace switch_cluster {
const function_generic_t *function_list = NULL;
const int function_flags = CLUSTER_FLAG_NONE;

cluster_t *create(endpoint_t *endpoint, config_t *config, uint8_t flags)
~~~
should be
~~~
namespace switch_cluster {
const function_generic_t *function_list = NULL;
const int function_flags = CLUSTER_FLAG_NONE;

cluster_t *create(endpoint_t *endpoint, config_t *config, uint8_t flags, uint32_t features)
~~~
So that I can set in the feature map when I create the cluster.

jadhavrohit924, 2025-01-02:
We will fix this.

jadhavrohit924, 2025-07-25:
This has been fixed, PTAL [RELEASE_NOTES](https://github.com/espressif/esp-matter/blob/main/RELEASE_NOTES.md#10-june-2025) and close this issue.