espressif/esp-matter #1224: Does ESP support running matter and BLE simultaneously?

Ajay2803, 2025-01-03:
Hello

I've utilized the example from https://github.com/oidebrett/starter-esp-matter-app, and we're planning to add BLE HID device control at the same time. I'm wondering if that's possible.

Here is an example of BLE HID that I have tested: https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/ble_hid_device_demo

I want to use my own app to control BLE devices by passing POSTMAN APIs to the matter device, which can then control and connect to BLE devices via HID proffile. Is it possible? And if so, how can I go about doing it?

I currently use ble exclusively for commissioning, after commissioning, BLE is deinilitiaized, so perhaps I can reinetialize ble and do this?

shubhamdp, 2025-01-03:
@Ajay2803 I'd suggest you to take a look at https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-16-using-ble-after-matter-commissioning

Ajay2803, 2025-01-17:
Hi,
Thank-you for your reply,

I have read this and now i am able to turn on BLE again after commissioning by doing gatt reset, there are not any kind of services running right now, But as i said before, i have tested **examples/bluetooth/bluedroid/ble/ble_hid_device_demo** this example and i want to do same with our device, can you guide me how can i do so to achieve same as above given example.

I have checked it the above given example is using bluedroid/ble and our matter device is using NimBLE so it is bit difficult for me to understand, if you can guide me that would be very helpful for me.

shubhamdp, 2025-03-18:
Hi @Ajay2803 you have [few open issues](https://github.com/espressif/esp-matter/issues/created_by/Ajay2803) related to Matter+Bluetooth, and I guess this is being discussed in some other issue. Can we close this one?

closed 2025-03-18 by Ajay2803