espressif/esp-matter #1225: How do i get IP of my matter device, in non matter app?

Ajay2803, 2025-01-03:
Hi,

I have made one project using esp-matter sdk, let's say matter light, it is working fine, and can be controlled through any matter app right?,
but i have my one custom app which is not matter supported, so i have implemented http server, using postman API's i am able to control relay or light in my custom app, but in my app implementation i have added all features of commissioning like scan qr share ssid and password of network and then it is connected to network and he got some ip, let say "192.168.23.333" using this i am able to control it and simultaneously on matter app through also.

But what if ip got changed, as in matter ip is assigned through dhcp, so it can change, so how can i get ip back in my custom app? is there any solution you may suggest? and i don't want to use static ip.

shubhamdp, 2025-01-06:
>can be controlled through any matter app right?

Light can be controlled by any Matter enabled controller.

IIUC, you are running the http server on ESP32, you have a custom phone/web/rest application to call APIs served by ESP32. Is my understanding correct?

> But what if ip got changed, as in matter ip is assigned through dhcp, 

Matter has nothing to do with the IPs. IPs are assigned to the device. Matter uses the mdns to find the device on the network, resolve those IPs and then use that to send the message to device.

If the device is commissioned using Matter, then it will have general diagnostic cluster and you can read the NetworkInterfaces attribute on that cluster. This cluster will be present on the root end point. This will give you the list of IP v4 and v6 addressed.

IMO, the right thing to do here is to use the mdns, device advertise the name and it can be resolved by your custom-app to get the ip-address to reach the device.

Ajay2803, 2025-01-07:
You are right, but what if we have multiple devices, then it will be difficult to identify our specific matter device because all have same mdns channel _matter.tcp something like this.

so i have found something called mdns channel delegate host name, it will assign a nick name to that channel, it will not change anything in current mdns and there is no need to read ip, we can use that delegate name instead of ip, but i have question can we change delegated name at run time more than once?

shubhamdp, 2025-01-10:
I think you can change the names at run-time.

wqx6, 2025-01-10:
Hi, A Matter device will identify itself with an unique instance name which is constructed from a 64-bit compressed Fabric identifier, and a 64-bit Node identifier. And you can resolve it by sending a mdns SRV query which includes the instance name, and the device will reply its host name and IP address(this is optional in SRV response) after it receives the query. If the IP address is not included in the response, then you can try to send an A/AAAA query to the host name to get the IP address. You can get more info about Operational device discovery in Matter SPEC 4.3.2. Operational Discovery.

jonsmirl, 2025-01-18:
@wqx6 the matter device browse API really should be extended to allow listing commissioned nodes in addition to un-commissioned ones. It is a minor change to the API.  

I have explained to over a dozen people how to write the code you describe above.  Since this keeps coming up over and over it really should be added to the core Matter API. It is also a tricky to write on some platforms because there are two things trying to us MDNS at the same time -- the matter code and the vendor code (who starts MDNS running?). This would be much easier if only one piece of code (Matter) managed MDNS.

wqx6, 2025-01-22:
> the matter device browse API really should be extended to allow listing commissioned nodes in addition to un-commissioned ones. It is a minor change to the API. For example, why can't a controller provide a list of commissioned nodes that it can see?

@jonsmirl In fact, Matter provides a console command which uses subtype filter to browse operational nodes in a specific fabric. You can try it with the console command `matter dns browse operational _I{compressed_fabric_id}`

jonsmirl, 2025-01-22:
I see, ResolverProxy::DiscoverOperationalNodes(DiscoveryFilter filter) is not exposed in the Android API. It would be useful it it was.

dhrishi, 2025-02-10:
@Ajay2803 Please close the issue if your question is answered

closed 2025-02-10 by Ajay2803