espressif/esp-matter #1226: Commissioning device with qr code fails

herculesp17, 2025-01-03:
**Describe the bug**
I am trying to commission the sensor matter example on my esp32c6, but the commissiong fails.

**Environment**
- ESP-Matter Commit Id:
96b09392313da50f5e8bce6cc0a52992b602834f
- ESP-IDF Commit Id:
67c1de1eebe095d554d281952fde63c16ee2dca0
- SoC (eg: ESP32 or ESP32-C3):
esp32-c6
- Device Logs (Please attach the log file):

- Host Machine OS:
EndeavourOS
- Host Machine Python version:
Python 3.12.7

**Further Description**
Steps to reproduce:

```bash
cd ~/esp/esp-matter/examples/sensors/
idf.py menuconfig
```

In the configuration menu, set the following additional configuration to use custom factory partition and different values for Data and Device Info Providers.

`Enable ESP32 Factory Data Provider [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Factory Data Provider]`

Enable config option CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER to use ESP32 specific implementation of CommissionableDataProvider and DeviceAttestationCredentialsProvider.

`Enable ESP32 Device Instance Info Provider [Component config → CHIP Device Layer → Commissioning options → Use ESP32 Device Instance Info Provider]`

Enable config option ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER to get device instance info from factory partition.

`Enable Attestation - Factory [ Component config → ESP Matter → DAC Provider options → Attestation - Factory]`

Enable config option CONFIG_FACTORY_PARTITION_DAC_PROVIDER to use DAC certificates from the factory partition during 

`Set chip-factory namespace partition label [Component config → CHIP Device Layer → Matter Manufacturing Options → chip-factory namespace partition label]`

Set config option CHIP_FACTORY_NAMESPACE_PARTITION_LABEL to choose the label of the partition to store key-values in the "chip-factory" namespace. The default chosen partition label is nvs.

### Generate a factory partition

```bash
esp-matter-mfg-tool -cn "My sensor" -v 0xFFF2 -p 0x8001 --pai \
    -k $MATTER_SDK_PATH/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
    -c $MATTER_SDK_PATH/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
    -cd $MATTER_SDK_PATH/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der
```

1. ```idf.py erase-flash```
2. ```idf.py flash -p /dev/ttyACM0```

```bash
esptool.py -p /dev/ttyACM0 write_flash 0x10000 out/fff2_8001/a0505a6b-1ec5-4454-8948-afa79ecb6ed8/a0505a6b-1ec5-4454-8948-afa79ecb6ed8-partition.bin
```

- Commissiong Logs (i ran idf.py monitor and the tried scanned the qr code for commissioning and these are the logs)

```
Detecting chip type... ESP32-C6
Running idf_monitor in directory /home/lennart/esp/esp-matter/examples/sensors
Executing "/home/lennart/.espressif/python_env/idf5.4_py3.12_env/bin/python /home/lennart/esp/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix riscv32-esp-elf- --target esp32c6 --revision 0 --decode-panic backtrace /home/lennart/esp/esp-matter/examples/sensors/build/sensors.elf -m '/home/lennart/.espressif/python_env/idf5.4_py3.12_env/bin/python' '/home/lennart/esp/esp-idf/tools/idf.py'"...
--- esp-idf-monitor 1.5.0 on /dev/ttyACM0 115200
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x15 (USB_UART_HPSYS),boot:0x1e (SPI_FAST_FLASH_BOOT)
[... 607 lines cut ...]
I (96570) app_main: Commissioning failed, fail safe timer expired
I (96580) chip[FS]: Fail-safe cleanly disarmed
I (96580) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (96590) NimBLE: GAP procedure initiated: advertise;
I (96590) NimBLE: disc_mode=2
I (96600) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (96600) NimBLE:

I (96610) chip[DL]: CHIPoBLE advertising started
I (96610) app_main: Fabric removed successfully
I (96620) esp_matter_core: BLE Disconnected
I (96620) chip[DL]: Configuring CHIPoBLE advertising (interval 25 ms, connectable)
I (96630) chip[DL]: Device already advertising, stop active advertisement and restart
I (96630) NimBLE: GAP procedure initiated: stop advertising.

I (96640) NimBLE: GAP procedure initiated: advertise;
I (96640) NimBLE: disc_mode=2
I (96650) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (96660) NimBLE:

I (100590) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000402's Attribute 0x00000000 is -4500 **********
I (100590) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000405's Attribute 0x00000000 is 0 **********
I (105590) esp_matter_attribute: ********** R : Endpoint 0x0001's Cluster 0x00000402's Attribute 0x00000000 is -4500 **********
I (105590) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000405's Attribute 0x00000000 is 0 **********
```

shubhamdp, 2025-01-06:
Can you please share the details about the commissioner app, hub, commissioner side logs, to get an idea where it failed.

After AddNOC step I see the fail safe timer is set to 0 by the commissioner, instead it should have started the discovery of the device on the mdns.

herculesp17, 2025-01-08:
I tried commissioning the device with the chip tool and found out what was wrong. 
The solution is in this topic espressif/esp-matter#514

It might be useful to update the README.md [here](https://github.com/espressif/esp-matter-tools/tree/main/mfg_tool#usage-examples), so that it is clear what the parameters are for generating a factory partition. I could help with that, if needed. The information could be added in the Configure you app section.

closed 2025-01-08 by herculesp17

shubhamdp, 2025-01-08:
@herculesp17 If you can, please raise the pull request with your suggestions.