espressif/esp-matter #1228: Fan slider attribute update fails

We-Inkling, 2025-01-04:
void update_matter_device_task(void *param)
{
    matter_device_t device;
    
    while (1) {
        // Wait for data to be available in the queue
        if (xQueueReceive(matterQueue, &device, pdMS_TO_TICKS(500))) {

            if (device.load != "Fan" && device.load != "Dimmer") {
                esp_matter_attr_val_t val = esp_matter_invalid(NULL);
                uint32_t attribute_id = OnOff::Attributes::OnOff::Id; 
                uint32_t cluster_id = OnOff::Id;
                attribute_t *attribute = attribute::get(device.id, cluster_id, attribute_id);
                attribute::get_val(attribute, &val);
                val.val.b = device.command;
                attribute::update(device.id, cluster_id, attribute_id, &val); 
            }
            else
            {
                if (device.load == "Fan") {

                    mode_flag = true;
                    percent_flag = true;
                    
                    printf("Fan Command: %d\n", device.command);
                    
                    uint32_t fan_cluster_id = FanControl::Id;
                    uint32_t attribute_id_mode = FanControl::Attributes::FanMode::Id;

                    esp_matter_attr_val_t val_m = esp_matter_invalid(NULL);
                    attribute_t *attribute_mode = attribute::get(device.id, fan_cluster_id, attribute_id_mode);
                    attribute::get_val(attribute_mode, &val_m);
                    if ( device.command > LOW_MODE_PERCENT_MIN && device.command <= LOW_MODE_PERCENT_MAX) val_m.val.u8 = chip::to_underlying(FanModeEnum::kLow);
                    else if ( device.command > MED_MODE_PERCENT_MIN && device.command <= MED_MODE_PERCENT_MAX) val_m.val.u8 = chip::to_underlying(FanModeEnum::kMedium);
                    else if ( device.command > HIGH_MODE_PERCENT_MIN) val_m.val.u8 = chip::to_underlying(FanModeEnum::kHigh);
                    else if (device.command == 0) val_m.val.u8 = chip::to_underlying(FanModeEnum::kOff);
                    update(device.id, fan_cluster_id, attribute_id_mode, &val_m);

                    
                    uint32_t attribute_id_percent = FanControl::Attributes::PercentSetting::Id; 
                    attribute_t *attribute_percent = attribute::get(device.id, fan_cluster_id, attribute_id_percent);
                    esp_matter_attr_val_t val_p = esp_matter_invalid(NULL);
                    attribute::get_val(attribute_percent, &val_p);
                    //printf("Device ID: %d, Cluster ID: %ld, Attribute ID: %ld\n", device.id, fan_cluster_id, attribute_id_mode);
                    val_p.val.u8 = device.command;
                    update(device.id, fan_cluster_id, attribute_id_percent, &val_p); 
                    
                    //printf("Device ID: %d, Cluster ID: %ld, Attribute ID: %ld\n", device.id, fan_cluster_id, attribute_id_percent);
                    
                    
                
                }
                else if (device.load == "Dimmer") {
                    esp_matter_attr_val_t val = esp_matter_invalid(NULL);
                    uint32_t attribute_id = LevelControl::Attributes::CurrentLevel::Id; 
                    uint32_t cluster_id = LevelControl::Id;
                    attribute_t *attribute = attribute::get(device.id, cluster_id, attribute_id);
                    get_val(attribute, &val);
                    val.val.u8 = device.command;
                    update(device.id,cluster_id,attribute_id, &val); 
                    dimmer_flag = true;
                }

            }
        }
    }
}

above is the code i run as a task to update the attributes of my various device types mainly light ,dimmers and fan when the respective touch are operated .
The seems to work fine for the dimmer and Light whereus in case of fan slider/percent/mode update sometimes it fails and sometime it works 

Please suggest a solution for this or alternate method to update the attributes fo the fan

jadhavrohit924, 2025-01-06:
@We-Inkling, are there any logs of when you failed to update an attribute? Logs will be beneficial in solving your issue.

We-Inkling, 2025-01-06:
Thank you for your response. I was able to identify and resolve the issue. The problem arose because the update was being executed as a FreeRTOS task, which caused it to proceed without waiting for the previous update to complete. Changing the implementation to a function call resolved the issue effectively.

Closing the issue.

closed 2025-01-06 by We-Inkling