espressif/esp-matter #1229: Cannot compile icd_app when targetting ESP32-H2

tomasmcguinness, 2025-01-05:
**Describe the bug**

I am trying to compile the icd_app example for the ESP32-H2. 

I execute this idf.py command:

```
idf.py -D SDKCONFIG_DEFAULTS="sdkconfig.defaults.esp32h2.lit" set-target esp32h2 build
```

but the compilation fails

```
./../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp: In member function 'CHIP_ERROR chip::DeviceLayer::ConnectivityManagerImpl::InitWiFi()':
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/CHIPDevicePlatformConfig.h:81:60: error: 'CONFIG_WIFI_STATION_RECONNECT_INTERVAL' was not declared in this scope; did you mean 'CONFIG_WIFI_AP_BEACON_INTERVAL'?
   81 | #define CHIP_DEVICE_CONFIG_WIFI_STATION_RECONNECT_INTERVAL CONFIG_WIFI_STATION_RECONNECT_INTERVAL
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp:393:67: note: in expansion of macro 'CHIP_DEVICE_CONFIG_WIFI_STATION_RECONNECT_INTERVAL'
  393 |     mWiFiStationReconnectInterval = System::Clock::Milliseconds32(CHIP_DEVICE_CONFIG_WIFI_STATION_RECONNECT_INTERVAL);
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/CHIPDevicePlatformConfig.h:81:60: note: maximum limit of 1000 namespaces searched for 'CONFIG_WIFI_STATION_RECONNECT_INTERVAL'
   81 | #define CHIP_DEVICE_CONFIG_WIFI_STATION_RECONNECT_INTERVAL CONFIG_WIFI_STATION_RECONNECT_INTERVAL
      |                                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp:393:67: note: in expansion of macro 'CHIP_DEVICE_CONFIG_WIFI_STATION_RECONNECT_INTERVAL'
  393 |     mWiFiStationReconnectInterval = System::Clock::Milliseconds32(CHIP_DEVICE_CONFIG_WIFI_STATION_RECONNECT_INTERVAL);
      |                                                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp:413:13: error: 'CONFIG_DEFAULT_WIFI_SSID' was not declared in this scope; did you mean 'CONFIG_ENABLE_WIFI_AP'?
  413 |         if (CONFIG_DEFAULT_WIFI_SSID[0] != 0)
      |             ^~~~~~~~~~~~~~~~~~~~~~~~
      |             CONFIG_ENABLE_WIFI_AP
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp:413:13: note: maximum limit of 1000 namespaces searched for 'CONFIG_DEFAULT_WIFI_SSID'
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp:422:45: error: 'CONFIG_DEFAULT_WIFI_PASSWORD' was not declared in this scope
  422 |             memcpy(wifiConfig.sta.password, CONFIG_DEFAULT_WIFI_PASSWORD,
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
../../../../../connectedhomeip/connectedhomeip/config/esp32/third_party/connectedhomeip/src/platform/ESP32/ConnectivityManagerImpl_WiFi.cpp:422:45: note: maximum limit of 1000 namespaces searched for 'CONFIG_DEFAULT_WIFI_PASSWORD'

```

Related I'm sure to the fact the H2 has no WiFi.

**Environment**
- ESP-Matter Commit Id: 96b0939
- ESP-IDF Commit Id: c9763f62dd
- SoC (eg: ESP32 or ESP32-C3): ESP32-H2
- Device Logs (Please attach the log file):

Full compilation output attached

[output.txt](https://github.com/user-attachments/files/18310727/output.txt)

wqx6, 2025-01-08:
Hi, I cannot reproduce this issue when I use the `sdkconfig.defaults.esp32h2.lit` to compile the example for ESP32-H2. Could you check whether the Wi-Fi is disabled in your sdkconfig file? https://github.com/espressif/esp-matter/blob/main/examples/icd_app/sdkconfig.defaults.esp32c6.lit#L20-L22. Looks like it was not disabled in your failing logs.

dhrishi, 2025-01-21:
@tomasmcguinness Can you please share any updates here? Please close the issue if resolved.

tomasmcguinness, 2025-01-21:
@dhrishi My sincere apologies for not updating this.

Yes, I did resolve the issue. I added the following to the bottom of the sdkconfig.defaults.esp32h2.lit file

```
CONFIG_ESP_WIFI_SOFTAP_SUPPORT=n
CONFIG_ENABLE_WIFI_AP=n
```

With those settings added, my original compile command worked successfully!

closed 2025-01-21 by tomasmcguinness

tomasmcguinness, 2025-01-22:
Could I raise a pull request with this fix?