espressif/esp-matter #1232: Unable to add device when using chip-tool

tien1608, 2025-01-06:
>>> pairing code-wifi 0x7283 Tien 888888889 23865233902  
[1736185466.751] [16820:16820] [TOO] Command: pairing code-wifi 0x7283 Tien 888888889 23865233902 
[1736185466.751] [16820:16854] [CTL] Setting wifi credentials from parameters
[1736185466.751] [16820:16854] [CTL] Setting attestation nonce to random value
[1736185466.751] [16820:16854] [CTL] Setting CSR nonce to random value
[1736185466.751] [16820:16854] [CTL] Stopping commissioning discovery over DNS-SD
[1736185466.751] [16820:16854] [CTL] Starting commissioning discovery over BLE
[1736185466.751] [16820:16854] [CTL] WiFi-PAF: has RendezvousInformationFlag::kWiFiPAF
[1736185466.751] [16820:16854] [CTL] Starting commissioning discovery over DNS-SD
[1736185466.752] [16820:16853] [BLE] BLE removing known devices
[1736185466.753] [16820:16853] [BLE] BLE initiating scan
[1736185466.867] [16820:16854] [BLE] ChipDeviceScanner has started scanning!
[1736185466.867] [16820:16854] [DL] Long dispatch time: 116 ms, for event type 3
[1736185467.012] [16820:16853] [BLE] New device scanned: C5:C2:F5:D5:42:DA
[1736185467.012] [16820:16853] [BLE] Device discriminator match. Attempting to connect.
[1736185467.072] [16820:16853] [BLE] ChipDeviceScanner has stopped scanning!
[1736185468.581] [16820:16853] [DL] ConnectDevice complete
[1736185468.581] [16820:16853] [BLE] New device connected: C5:C2:F5:D5:42:DA
[1736185469.547] [16820:16853] [DL] CHIP service found
[1736185469.547] [16820:16853] [DL] Valid C1 characteristic found
[1736185469.547] [16820:16853] [DL] Valid C2 characteristic found
[1736185469.547] [16820:16853] [DL] New BLE connection: conn=0x7a742801c7c0 device=C5:C2:F5:D5:42:DA path=/org/bluez/hci0/dev_C5_C2_F5_D5_42_DA
[1736185469.547] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16387
[1736185469.547] [16820:16854] [CTL] Discovered device to be commissioned over BLE
[1736185469.547] [16820:16854] [CTL] Attempting PASE connection to BLE
[1736185469.547] [16820:16854] [IN] BleConnectionComplete: endPoint 0x60e2d8b90ba0
[1736185469.547] [16820:16854] [IN] SecureSession[0x7a7438040640]: Allocated Type:1 LSID:26918
[1736185469.547] [16820:16854] [SC] Assigned local session key ID 26918
[1736185469.547] [16820:16854] [EM] <<< [E:5304i S:0 M:111525882] (U) Msg TX from 0C310652880784D5 to 0:0000000000000000 [0000] [BLE] --- Type 0000:20 (SecureChannel:PBKDFParamRequest) (B:98)
[1736185469.548] [16820:16854] [IN] Message appended to BLE send queue
[1736185469.548] [16820:16854] [SC] Sent PBKDF param request [II:500ms AI:300ms AT:4000ms)
[1736185469.548] [16820:16854] [CTL] Commission called for node ID 0x0000000000007283
[1736185469.976] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16389
[1736185470.066] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16390
[1736185470.066] [16820:16854] [BLE] subscribe complete, ep = 0x60e2d8b90ba0
[1736185470.067] [16820:16853] [DL] Indication received, conn = 0x7a742801c7c0
[1736185470.067] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16391
[1736185470.067] [16820:16854] [BLE] peripheral chose BTP version 4; central expected between 4 and 4
[1736185470.067] [16820:16854] [BLE] using BTP fragment sizes rx 244 / tx 244.
[1736185470.067] [16820:16854] [BLE] local and remote recv window size = 5
[1736185470.067] [16820:16854] [IN] BLE EndPoint 0x60e2d8b90ba0 Connection Complete
[1736185470.156] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16389
[1736185470.160] [16820:16853] [DL] Indication received, conn = 0x7a742801c7c0
[1736185470.160] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16391
[1736185470.160] [16820:16854] [EM] >>> [E:5304i S:0 M:130029436] (U) Msg RX from 0:0000000000000000 [0000] to 0C310652880784D5 --- Type 0000:21 (SecureChannel:PBKDFParamResponse) (B:170)
[1736185470.161] [16820:16854] [EM] Found matching exchange: 5304i, Delegate: 0x7a7438043218
[1736185470.161] [16820:16854] [SC] Received PBKDF param response
[1736185470.161] [16820:16854] [SC] Peer assigned session ID 51316
[1736185470.161] [16820:16854] [SC] Found MRP parameters in the message
[1736185470.188] [16820:16854] [EM] <<< [E:5304i S:0 M:111525883] (U) Msg TX from 0C310652880784D5 to 0:0000000000000000 [0000] [BLE] --- Type 0000:22 (SecureChannel:PASE_Pake1) (B:92)
[1736185470.188] [16820:16854] [SC] Sent spake2p msg1
[1736185470.292] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16389
[1736185470.834] [16820:16853] [DL] Indication received, conn = 0x7a742801c7c0
[1736185470.834] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16391
[1736185470.835] [16820:16854] [EM] >>> [E:5304i S:0 M:130029437] (U) Msg RX from 0:0000000000000000 [0000] to 0C310652880784D5 --- Type 0000:23 (SecureChannel:PASE_Pake2) (B:127)
[1736185470.835] [16820:16854] [EM] Found matching exchange: 5304i, Delegate: 0x7a7438043218
[1736185470.835] [16820:16854] [SC] Received spake2p msg2
[1736185470.835] [16820:16854] [EM] <<< [E:5304i S:0 M:111525884] (U) Msg TX from 0C310652880784D5 to 0:0000000000000000 [0000] [BLE] --- Type 0000:24 (SecureChannel:PASE_Pake3) (B:59)
[1736185470.836] [16820:16854] [SC] Sent spake2p msg3
[1736185470.922] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16389
[1736185470.922] [16820:16853] [DL] Indication received, conn = 0x7a742801c7c0
[1736185470.922] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16391
[1736185470.922] [16820:16854] [EM] >>> [E:5304i S:0 M:130029438] (U) Msg RX from 0:0000000000000000 [0000] to 0C310652880784D5 --- Type 0000:40 (SecureChannel:StatusReport) (B:30)
[1736185470.922] [16820:16854] [EM] Found matching exchange: 5304i, Delegate: 0x7a7438043218
[1736185470.923] [16820:16854] [SC] SecureSession[0x7a7438040640, LSID:26918]: State change 'kEstablishing' --> 'kActive'
[1736185470.923] [16820:16854] [IN] SecureSession[0x7a7438040640]: Activated - Type:1 LSID:26918
[1736185470.923] [16820:16854] [IN] New secure session activated for device <FFFFFFFB00000000, 0>, LSID:26918 PSID:51316!
[1736185470.923] [16820:16854] [CTL] Remote device completed SPAKE2+ handshake
[1736185470.923] [16820:16854] [CTL] PASE session established with commissionee. Stopping discovery.
[1736185470.923] [16820:16854] [CTL] Stopping commissioning discovery over DNS-SD
[... 608 lines cut ...]
[1736185472.365] [16820:16854] [DMG] 	],
[1736185472.365] [16820:16854] [DMG] 	
[1736185472.365] [16820:16854] [DMG] 	InteractionModelRevision = 11
[1736185472.365] [16820:16854] [DMG] },
[1736185472.365] [16820:16854] [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0001
[1736185472.365] [16820:16854] [CTL] Failsafe disarmed
[1736185472.365] [16820:16854] [CTL] Successfully finished commissioning step 'Cleanup'
[1736185472.365] [16820:16854] [DIS] Closing all BLE connections
[1736185472.365] [16820:16854] [IN] Clearing BLE pending packets.
[1736185472.365] [16820:16854] [BLE] Auto-closing end point's BLE connection.
[1736185472.365] [16820:16854] [DL] Closing BLE GATT connection (con 0x7a742801c7c0)
[1736185472.365] [16820:16853] [DL] Close BLE connection: peer=C5:C2:F5:D5:42:DA
[1736185472.405] [16820:16854] [IN] SecureSession[0x7a7438040640]: MarkForEviction Type:1 LSID:26918
[1736185472.405] [16820:16854] [SC] SecureSession[0x7a7438040640, LSID:26918]: State change 'kActive' --> 'kPendingEviction'
[1736185472.405] [16820:16854] [CTL] Commissioning complete for node ID 0x0000000000007283: src/controller/CHIPDeviceController.cpp:1332: CHIP Error 0x000000AC: Internal error
[1736185472.405] [16820:16854] [TOO] Device commissioning Failure: src/controller/CHIPDeviceController.cpp:1332: CHIP Error 0x000000AC: Internal error
[1736185472.405] [16820:16854] [TOO] Run command failure: src/controller/CHIPDeviceController.cpp:1332: CHIP Error 0x000000AC: Internal error
[1736185472.406] [16820:16854] [DMG] ICR moving to [AwaitingDe]
[1736185472.406] [16820:16854] [IN] SecureSession[0x7a7438040640]: Released - Type:1 LSID:26918
[1736185472.406] [16820:16854] [DL] HandlePlatformSpecificBLEEvent 16390
[1736185472.406] [16820:16854] [BLE] No endpoint for unsubscribe complete
[1736185472.406] [16820:16853] [DL] BLE connection closed: conn=0x7a742801c7c0
[1736185472.406] [16820:16854] [BLE] No endpoint for connection error

i diy a switch device using esp32c3 with esp zero code. i tried to operate it with chip-tool but got this error. i don't know how to fix it.

shubhamdp, 2025-01-07:
@tien1608 ESP-Zerocode uses the test-net PAA uploaded by Espressif, and hence you are getting attestation error:101.

```
[1736185472.287] [16820:16854] [CTL] Failed in verifying 'Attestation Information' command received from the device: err 101. Look at AttestationVerificationResult enum to understand the errors
```

You will need to download the Espressif's test net PAA and provide that to chip-tool using `--paa-trust-store-path` argument.

Since you are using chip-tool, that means you have a clone of connectedhomeip repository. Please follow below steps to fetch and use the test-net PAAs
```
cd connectedhomeip
# OR --> cd esp-matter/connectedhomeip/connectedhomeip (if using esp-matter)

cd credentials
python3 python3 ./fetch_paa_certs_from_dcl.py --use-test-net-http --paa-trust-store-path test-net-paa-certs
```

When using with chip-tool, use the path with `--paa-trust-store-path` argument, eg:
```
./chip-tool pairing code 1 12921148873 --paa-trust-store-path /Users/shubhampatil/Downloads/certs./chip-tool pairing code 1 12921148873 --paa-trust-store-path /path/to/test-net-paa-certs
```

tien1608, 2025-01-07:
@shubhamdp I installed test-net-paa-certs but when I run this command I still get the same problem:
pairing code-wifi 1 Tien 888888889 05332742594 --paa-trust-store-path $HOME/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test-net-paa-certs

shubhamdp, 2025-01-09:
Can you share the logs from device and chip-tool, please do not paste them, attach the log file.

tien1608, 2025-01-09:
> Can you share the logs from device and chip-tool, please do not paste them, attach the log file.

@shubhamdp 
Below is the log file from my chip-tool and device console.
[chip_tool.txt](https://github.com/user-attachments/files/18358256/chip_tool.txt)
[device_console.txt](https://github.com/user-attachments/files/18358257/device_console.txt)

shubhamdp, 2025-01-09:
Can you try this outside the interactive mode, like below?
```
./chip-tool pairing code 1 12921148873 --paa-trust-store-path  /path/to/test-net-paa-certs
```

Path set at the beginning will be used instead of path provided in interactive mode. i.e. to the `chip-tool` command.

Or export path into `CHIPTOOL_PAA_TRUST_STORE_PATH`
```
export CHIPTOOL_PAA_TRUST_STORE_PATH=/path/to/test-net-paa-certs
```

Source code for which PAAs to use can be found at: https://github.com/project-chip/connectedhomeip/blob/f8d457a446456d5df1e750e728ea48e8fce9d989/examples/chip-tool/commands/common/CHIPCommand.cpp#L64

tien1608, 2025-01-09:
> Can you try this outside the interactive mode, like below?
>
> ```
> ./chip-tool pairing code 1 12921148873 --paa-trust-store-path  /path/to/test-net-paa-certs
> ```
>
> Path set at the beginning will be used [...]

@shubhamdp Thank you very much, I have successfully connected the bulb device to the chip-tool. But I have a question: why can't I find the device when in interactive mode?
[not_interactive.txt](https://github.com/user-attachments/files/18359244/not_interactive.txt)
[interactive.txt](https://github.com/user-attachments/files/18359248/interactive.txt)

tien1608, 2025-01-09:
@shubhamdp Thank you very much, my problem is solved.

closed 2025-01-09 by tien1608