espressif/esp-matter #1233: Trying to use TimeSynchronizationCluster

bnadim, 2025-01-07:
**Describe the question/query that you have**
Hi, i am currently trying to use the TimeSynchronizationCluster to get time.
I am using a device connected to a HomePod which should support time synchronization.

So far i have tried to add this code based on light switch example: 
```
#ifdef CONFIG_ENABLE_SNTP_TIME_SYNC
    endpoint_t *root_node_ep = endpoint::get_first(node);
    ABORT_APP_ON_FAILURE(root_node_ep != nullptr, ESP_LOGE(TAG, "Failed to find root node endpoint"));

    cluster::time_synchronization::config_t time_sync_cfg;
    static chip::app::Clusters::TimeSynchronization::DefaultTimeSyncDelegate time_sync_delegate;
    time_sync_cfg.delegate = &time_sync_delegate;
    time_sync_cluster = cluster::time_synchronization::create(root_node_ep, &time_sync_cfg, CLUSTER_FLAG_SERVER | CLUSTER_FLAG_CLIENT);
    ABORT_APP_ON_FAILURE(time_sync_cluster != nullptr, ESP_LOGE(TAG, "Failed to create time_sync_cluster"));

    cluster::time_synchronization::feature::time_sync_client::add(time_sync_cluster);
    /*cluster::time_synchronization::feature::time_zone::config_t tz_cfg;
    cluster::time_synchronization::feature::time_zone::add(time_sync_cluster, &tz_cfg);*/
#endif
```

And i try to read the value using: 
```
attribute_t *utc_time = attribute::get(time_sync_cluster,TimeSynchronization::Attributes::UTCTime::Id);
    if (utc_time == nullptr) {
        ESP_LOGI(TAG, "utc_time attribute is null");
    } else {
        esp_matter_attr_val_t val = esp_matter_invalid(nullptr);
        attribute::get_val(utc_time, &val);

        ESP_LOGI(TAG, "TIME IS: %lli", val.val.u64);
    }
```

which throw an error ```esp_matter_core: get_val(1144): Attribute is not managed by esp matter data model```

So i am pretty sure i am not using properly. 
I wanted to use CLUSTER_FLAG_CLIENT flag only on the cluster, but i saw in code that it doesn't create in this case utc_time and granularity attributes which are mandatory based on matter spec 1.3.

Who can i get synchronized time ? I think it's missing an example.

Thank you for your kind help.

**Additional context**
...

wqx6, 2025-01-08:
The UTCTime attribute is managed by AttributeAccessInterface instead of esp-matter data model so you cannot get its value with the APIs provided in esp-matter component, Please refer to [this](https://github.com/espressif/connectedhomeip/blob/43aa98c2d30ee547c6b587b9de7bbb794f175ece/src/app/clusters/time-synchronization-server/time-synchronization-server.cpp#L1040-L1052), the unix UTC time can be get `System::SystemClock().GetClock_RealTime()`.

bnadim, 2025-01-10:
Hey wqx6, 

Thank you for you feedback. I tried, but my time doesn't get synced. I can't find what i am missing. Do you have a "full example" please ? I believe it would make a very good example. 
Thank you very much for your help.

Regards

jonsmirl, 2025-01-14:
Using this cluster is very painful. Just turn on the standard NTP client via menuconfig which is 1000 times easier.

wqx6, 2025-01-16:
> Thank you for you feedback. I tried, but my time doesn't get synced. I can't find what i am missing. Do you have a "full example" please ? I believe it would make a very good example.
Thank you very much for your help.

What commissioner are you using? I tested the time-sync cluster with the following steps:
- Commission the device with chip-tool and then open a commissioning window.
- Use Apple home app to commission the device, the Apple home app will set the UTC time and time zone during commissioning
- Read the UTC time with chip-tool.

bnadim, 2025-01-17:
Hey, 

thank you for your feedback. @wqx6 indeed i finally managed to get time at commissioning. My issue is after restart i lose time and i didn't find to request it unless i recommission the device. Apple set the utc time at commissioning but not trusted source. 
Do you have any idea on how to "continuously" sync with an apple device ? 

@jonsmirl i tried standard ntp, but it doesn't get sync if connected to thread/matter network of apple. I juste get timeouts. Have you done it with apple ? 

Thank you

jonsmirl, 2025-01-18:
Standard NTP will ignore Matter and use the existing NTP protocol. Use pool.ntp.org which will attach to the nearest public server. Of course this only works from a network with Internet connectivity. If you have an Apple Home device you have have Internet connectivity since it also needs it.  

This might be a thread problem, does thread route onto the Internet by default?  For sure it is possible for Thread to route onto the Internet but I don't know if that is capability is turned on. All of the Matter devices I work with are wifi based.

jonsmirl, 2025-01-18:
Since thread is IPv6 only I believe you need to use 2.pool.ntp.org

caipiblack, 2025-02-06:
Hello,

If you need help to use the time synchronization cluster, take a look here: https://github.com/espressif/esp-matter/issues/1269

I made it working.

There is just a little bug but it is not very important:
- Most of the logs display the correct date/time
- One line of logs display the wrong date/time
- In the code when I use `chip::System::SystemClock().GetClock_RealTime()` i get the correct time.