espressif/esp-matter #1235: TC_DESC_2_2: Ensure each of the listed endpoints has a tag attribute and the tag attributes are not the same

bilalahmaddev, 2025-01-10:
**Describe the bug**
 Ensure each of the listed endpoints has a tag attribute and the tag attributes are not the same
**Environment**
- ESP-Matter Commit Id: relase1.3
- ESP-IDF Commit Id: 5.2.1
- SoC (eg: ESP32 or ESP32-C3): ESP32-S3
- Device Logs (Please attach the log file):
[TC-DESC-2-2_2025_01_10_14_15_04.log](https://github.com/user-attachments/files/18378413/TC-DESC-2-2_2025_01_10_14_15_04.log)

- Host Machine OS:
- Host Machine Python version:

- Commissioner app and versions if present: certification tool  Version: 2.10.2+spring2024
- Home hub app and versions if present:
- Commissioner's logs if present:

bilalahmaddev, 2025-01-13:
@dhrishi @VaishaliAvhale can you please look into this and others I created they are certification blockers for us.

jonsmirl, 2025-01-14:
Multiple identical endpoints now have to be differentiated via TAGs in the descriptor.  This requires a enabling the TAGLIST feature on the Descriptor cluster.

https://github.com/espressif/esp-matter/issues/1141

bilalahmaddev, 2025-01-14:
@jonsmirl Thanks. From where I can find namespaces required for TagList, link mentioned in specification is not public:
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/namespaces

bilalahmaddev, 2025-01-14:
Got it there is a document for it in specification documents.

bilalahmaddev, 2025-01-14:
which namespace should be used for endpoints of type on off plugin and dimmable plugin?

jonsmirl, 2025-01-14:
I don't think it matters, they just have to be different.

bilalahmaddev, 2025-01-15:
@jonsmirl is this a correct way to add Tag to each endpoint descriptive cluster TagList:

const Descriptor::Structs::SemanticTagStruct::Type ep0TagList[] = {
    {.namespaceID = 0x00,
     .tag = 0x00}};

const Descriptor::Structs::SemanticTagStruct::Type ep1TagList[] = {
    {.namespaceID = 0x01,
     .tag = 0x01}};

did this for each endpoint
 
cluster_t* descriptor = cluster::get(output_channels_endpoint, Descriptor::Id);
 descriptor::feature::taglist::add(descriptor);
  
  
  SetTagList(0, chip::Span<const Descriptor::Structs::SemanticTagStruct::Type>(ep0TagList));
  SetTagList(1, chip::Span<const Descriptor::Structs::SemanticTagStruct::Type>(ep1TagList));

I am not getting any cluster id in logs: 

[image]

bilalahmaddev, 2025-01-15:
My bad I was calling SetTagList() before matter started. closing as issue is fixed.

closed 2025-01-15 by bilalahmaddev