espressif/esp-matter #1236: TC_RR_1_1.py: MaxGroupsPerFabric value: 4, number of endpoints with Groups clusters cluster: 12

bilalahmaddev, 2025-01-11:
**Describe the bug**

INFO MaxGroupsPerFabric value: 4, number of endpoints with Groups clusters cluster: 12, which are: [10, 1, 9, 4, 6, 3, 12, 7, 2, 8, 11, 5]

**Environment**
- ESP-Matter Commit Id: matter release/1.3
- ESP-IDF Commit Id: 5.2.1
- SoC (eg: ESP32 or ESP32-C3): esp32=s3
- Device Logs (Please attach the log file):

[TC_RR_1_1.txt](https://github.com/user-attachments/files/18385181/TC_RR_1_1.txt)

- Commissioner app and versions if present:
- Home hub app and versions if present:

jonsmirl, 2025-01-14:
This is a new test which a lot of old devices fail. I think the error is saying you have 12 endpoints and MaxGroupsPerFabric is set to 4. You have to change this to 4*12 = 48.

I have a 6-button Zemismart switch here with MaxGroupsPerFabric set to 4 and it has six buttons. How can six buttons possibly work with only 4 groups enabled? The switch shipped before this test was added.

bilalahmaddev, 2025-01-14:
@jonsmirl I am able to resolve it by setting CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC in CHIPConfig.h from 1 to 12.

shubhamdp, 2025-01-14:
To resolve the dependency on this config we did https://github.com/espressif/esp-matter/commit/974997de95dba6478b56cfba40f71fd59e0ba06d which counts the number of endpoints supporting cluster. But this change is not a part of release/v1.3

Let me know if the above works for you else I can provide the patch on release/v1.3 and also get it merged into release/v1.3

bilalahmaddev, 2025-01-14:
@shubhamdp getting this error after checkout out your commit:
[image]

shubhamdp, 2025-01-14:
@bilalahmaddev did you checked out that specific commit and reset to that?  I'd suggest you cherry-pick that one and just re-build the firmware...
```
cd esp-matter
# expecting this to be latest release/v1.3 as per your description
git cherry-pick 974997de95dba6478b56cfba40f71fd59e0ba06d 
cd to-example
idf.py build
```

I tried and was able to build the light successfully.

bilalahmaddev, 2025-01-14:
@shubhamdp this change introduces another issue: 
[image]

bilalahmaddev, 2025-01-14:
@shubhamdp I was calling this API earlier to perform factory reset:

chip::Server::GetInstance().ScheduleFactoryReset();

but after running TR-1.1 test, this was not working as expected and getting above mentioned issue. But calling this API directly resolve the issue:

ConfigurationMgr().InitiateFactoryReset();

bilalahmaddev, 2025-01-14:
@shubhamdp 
[image]

bilalahmaddev, 2025-01-14:
@shubhamdp some how ground count is set to 35 and if I use 8 endpoints test passes but if I use anything above 8 it fails like in this case it failed when I used 9 endpoits, we need 12 endpoints. can you please help me fix this? Our device is already with ATL we need to send fix of this test to them. 
mobly.signals.TestFailure: Details=35 != 36 Found group count does not match written value., Extras=None
[image]

bilalahmaddev, 2025-01-15:
@shubhamdp @jonsmirl can you please help me with this?

shubhamdp, 2025-01-15:
@bilalahmaddev when using esp-matter you should ideally use [`esp_matter::factory_reset()`](https://github.com/espressif/esp-matter/blob/main/components/esp_matter/esp_matter_core.h#L75) which deletes required matter data added by esp-matter as well as connectedhomeip SDK.

`chip::Server::GetInstance().ScheduleFactoryReset();` only deletes the connectedhomeip's non-volatile data.

Also `ScheduleFactoryReset()` calls `ConfigurationMgr().InitiateFactoryReset();` please see: https://github.com/project-chip/connectedhomeip/blob/master/src/app/server/Server.cpp#L625

shubhamdp, 2025-01-15:
Since you have (1 + 12) endpoint, NVS size needs to be increased, we are still figuring out why it is failing with 12 endpoints.

The latest error that we are seeing is 
```
******************************************************************
*
* Test test_TC_RR_1_1 failed for the following reason:
* Details=35 != 48 Found group count does not match written value., Extras=None
*
* File "/Users/shubhampatil/esp/connectedhomeip/src/python_testing/TC_RR_1_1.py", line 714, in validate_group_table
* asserts.assert_equal(found_groups, len(group_table_written[client_idx]),
*
* Test step:
*     UNKNOWN - no test steps provided in test script
*
* Endpoint: None
*
*******************************************************************
```

bilalahmaddev, 2025-01-15:
@shubhamdp Yes I am also getting same issue when (12+1) endpoints are used regardless of how much nvs size I set ( even tried by increasing to 512kb) . This test cases passe up to (8+1) endpoints only after increasing nvs size because this 35 is less than 32 (8x4) which is required for per group per endpoint. Please see the screenshots I shared above. Somehow device is not storing group key map for more than 8 endpoints. 

I will be thankful if you can provide solution as soon as possible. Let me know if you need anything from my side.

crabshack-offical, 2025-01-16:
@shubhamdp we are having same issue and have our laboratory testing held up because of one failure which is this issue. What is status of resolving please?

bilalahmaddev, 2025-01-16:
@shubhamdp any update on this?

shubhamdp, 2025-01-16:
@bilalahmaddev finally after a lot of bit-banging here and there, we narrowed down that issue could have been with the chunking. Since 48 entries would be a lot bigger so needs to be chunked and there was a bug in encoding group key management's attributes.

And it was fixed in https://github.com/project-chip/connectedhomeip/commit/14b332966cde23ea425e99ebfe5dcd7ef7dd076d.

Can you please update the submodule to latest v1.3-branch and give it a try...

```
// This is for automatically counting the groups endpoint count   // NOTE: you may already have this one
cd esp-matter
# expecting this to be latest release/v1.3 as per your description
git cherry-pick 974997de95dba6478b56cfba40f71fd59e0ba06d 

// pull the latest v1.3-branch which has fix
cd connectedhomeip/connectedhomeip
git fetch origin v1.3-branch
git reset --hard FETCH_HEAD

cd to-example
idf.py build // and then run the TC_RR_1_1
```

bilalahmaddev, 2025-01-17:
@shubhamdp Thanks for this. I am trying now, will update you shortly.

bilalahmaddev, 2025-01-17:
@shubhamdp Yes testing is passing now. But I think we will have to send a new device to ATL because increase in nvs size required to adjust 48 entries in table. We can not perform OTA of partition table. Right?

shubhamdp, 2025-01-17:
@bilalahmaddev yes we cannot OTA the partition table.

bilalahmaddev, 2025-01-17:
@shubhamdp when will this https://github.com/espressif/esp-matter/commit/974997de95dba6478b56cfba40f71fd59e0ba06d merged into release/1.3?

shubhamdp, 2025-01-17:
@bilalahmaddev this will be out early next week.

shubhamdp, 2025-01-22:
@bilalahmaddev we have cherry-picked the 974997de95dba6478b56cfba40f71fd59e0ba06d, and bumped the submodule to fix the groups cluster related ƒailure. Please fetch the latest release/v1.3 branch to include them.

Also, I think https://github.com/espressif/esp-matter/issues/1240 is related to same test case.

If all the failures regaridng `TC_RR_1_1` test case are fixed. Please close the issues.

bilalahmaddev, 2025-01-22:
Yes https://github.com/espressif/esp-matter/issues/1240 is also related to this. Closing both. Thanks for your support @shubhamdp

closed 2025-01-22 by bilalahmaddev