espressif/esp-matter #1239: _DoubleExceptionVector

jonsmirl, 2025-01-12:
I was not using the debugger when I hit this:
ESP-IDF v5.2.3

~~~
cluBLE 8 d4 ae f1 34 a4 
cluBLE 8 d4 ae f1 34 a4 
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x403743c0  PS      : 0x00040536  A0      : 0x00000000  A1      : 0x3fcc55b0  
--- 0x403743c0: _DoubleExceptionVector at /home/jonsmirl/esp/esp-idf/components/xtensa/xtensa_vectors.S:564

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3fca531c  
A6      : 0x00000001  A7      : 0x3fcc5754  A8      : 0x821b25e1  A9      : 0x3fcc5590  
A10     : 0x00060023  A11     : 0x00000003  A12     : 0x00000000  A13     : 0x00000008  
A14     : 0x00000001  A15     : 0x3fca50a8  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
EXCVADDR: 0xffffffe0  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x403743bd:0x3fcc55b0
--- 0x403743bd: _UserExceptionVector at ??:?

ELF file SHA256: 7ac7a9cfbb8a5ad0

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x29 (SPI_FAST_FLASH_BOOT)
Saved PC:0x42156ce8
--- 0x42156ce8: usb_serial_jtag_ll_ena_intr_mask at /home/jonsmirl/esp/esp-idf/components/hal/esp32s3/include/hal/usb_serial_jtag_ll.h:50
 (inlined by) usb_serial_jtag_write_bytes at /home/jonsmirl/esp/esp-idf/components/driver/usb_serial_jtag/usb_serial_jtag.c:228

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x19fc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xdd0
load:0x403cc700,len:0x3120
entry 0x403c992c

~~~

jonsmirl, 2025-01-13:
Another variation on same fault
~~~
log_handler3
Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  1 register dump:
PC      : 0x403743c0  PS      : 0x00040236  A0      : 0x00000000  A1      : 0x3fcc55b0  
--- 0x403743c0: _DoubleExceptionVector at /home/jonsmirl/esp/esp-idf/components/xtensa/xtensa_vectors.S:564

A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3fca531c  
A6      : 0x00000001  A7      : 0x3fcc5754  A8      : 0x821b261d  A9      : 0x3fcc5590  
A10     : 0x00060023  A11     : 0x00000003  A12     : 0x00000000  A13     : 0x00000008  
A14     : 0x00000001  A15     : 0x3fca50a8  SAR     : 0x00000000  EXCCAUSE: 0x00000001  
EXCVADDR: 0xffffffe0  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x403743bd:0x3fcc55b0
--- 0x403743bd: _UserExceptionVector at ??:?

ELF file SHA256: c471ed8de2220af2

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x403848c6
--- 0x403848c6: xPortEnterCriticalTimeout at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:488

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3820,len:0x19fc
load:0x403c9700,len:0x4
load:0x403c9704,len:0xdd0
load:0x403cc700,len
~~~

jonsmirl, 2025-01-14:
I am still testing, but my preliminary assessment is that this is due to the HTTP server overrunning the FreeRTOS timer stack.

jonsmirl, 2025-01-14:
This is very suspicious, is that NVS set causing the timer stack to overflow?  I set my timer stack size up to 10K and I am rerunning the test.
~~~
I (3601073) chip[DL]: NVS set: chip-counters/total-hours = 316 (0x13C)

***ERROR*** A stack overflow in task Tmr Svc has been detected.

Backtrace: 0x40375dd6:0x3fcc5b60 0x40384399:0x3fcc5b80 0x403849b1:0x3fcc5ba0 0x4038525a:0x3fcc5c20 0x40384498:0x3fcc5c50 0x4038448e:0x00060023 |<-CORRUPTED
--- 0x40375dd6: panic_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/panic.c:466
0x40384399: esp_system_abort at /home/jonsmirl/esp/esp-idf/components/esp_system/port/esp_system_chip.c:93
0x403849b1: vApplicationStackOverflowHook at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:558
0x4038525a: vTaskSwitchContext at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/tasks.c:3664 (discriminator 7)
0x40384498: _frxt_dispatch at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:451
0x4038448e: _frxt_int_exit at /home/jonsmirl/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/portasm.S:246

~~~

jonsmirl, 2025-01-14:
This is an old error which I have hit previously. You can't do ChipLogProgress() from an interrupt routine; it causes a stack overflow. I commented out the NVS Set debug messages in this file:
https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32Config.cpp

Well.... you can do it, but then I have to make 10K timer stacks.

I (3601073) chip[DL]: NVS set: chip-counters/total-hours = 316 (0x13C)
***ERROR*** A stack overflow in task Tmr Svc has been detected.

That timer service routine has to switch off from the FreeRTOS timer stack and onto the CHIP stack.

shubhamdp, 2025-01-14:
@jonsmirl Thanks for raising and drilling down on the problem. I did not hit this during my tests so I assumed the logs would go-through.

Since total-operational-hours is a critical information, it intentionally uses the FreeRTOS timer to increment the values, so that it should not be a victim of PostEvent failures.

Here is the PR for the fix: https://github.com/project-chip/connectedhomeip/pull/37058

jonsmirl, 2025-01-14:
You can print with ESP_LOG you just can't use ChipLogProgress since it is a stack pig.

shubhamdp, 2025-03-18:
@jonsmirl The pull request https://github.com/project-chip/connectedhomeip/pull/37058 has been merged, and the submodule has been updated. The latest main branch now includes this change, so this issue can be closed.

jonsmirl, 2025-03-18:
ok

closed 2025-03-18 by jonsmirl