espressif/esp-matter #1240: TC_RR_1.1 : ERROR Time out! failed to receive write response from Exchange: 18859i

bilalahmaddev, 2025-01-12:
**Describe the bug**
Channel:StandaloneAck)
[MatterTest] 01-12 18:28:44.531 INFO Found an existing secure session to [1:0000000012344321]!
[MatterTest] 01-12 18:28:44.550 INFO <<< [E:18857i S:42526 M:72266529] (S) Msg TX to 1:0000000012344321 [7609] [UDP:[fe80::8265:99ff:fec3:cf64%eth0]:5540] --- Type 0001:06 (IM:WriteRequest)
[MatterTest] 01-12 18:28:44.926 INFO Retransmitting MessageCounter:72266529 on exchange 18857i Send Cnt 1
[MatterTest] 01-12 18:28:45.269 INFO Retransmitting MessageCounter:72266529 on exchange 18857i Send Cnt 2
[MatterTest] 01-12 18:28:45.801 INFO Retransmitting MessageCounter:72266529 on exchange 18857i Send Cnt 3
[MatterTest] 01-12 18:28:46.797 INFO Retransmitting MessageCounter:72266529 on exchange 18857i Send Cnt 4
[MatterTest] 01-12 18:28:48.261 ERROR Failed to Send CHIP MessageCounter:72266529 on exchange 18857i sendCount: 4 max retries: 4
[MatterTest] 01-12 18:28:50.279 INFO >>> [E:18857i S:42526 M:185962603 (Ack:72266529)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0001:07 (IM:WriteResponse)
[MatterTest] 01-12 18:28:50.365 INFO <<< [E:18857i S:42526 M:72266530 (Ack:185962603)] (S) Msg TX to 1:0000000012344321 [7609] [UDP:[fe80::8265:99ff:fec3:cf64%eth0]:5540] --- Type 0001:06 (IM:WriteRequest)
[MatterTest] 01-12 18:28:50.367 INFO >>> [E:18857i S:42526 M:185962604 (Ack:72266529)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:50.368 INFO >>> [E:18857i S:42526 M:185962605 (Ack:72266529)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:50.378 INFO >>> [E:18857i S:42526 M:185962606 (Ack:72266529)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:50.395 INFO >>> [E:18857i S:42526 M:185962607 (Ack:72266529)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:50.742 INFO Retransmitting MessageCounter:72266530 on exchange 18857i Send Cnt 1
[MatterTest] 01-12 18:28:51.109 INFO Retransmitting MessageCounter:72266530 on exchange 18857i Send Cnt 2
[MatterTest] 01-12 18:28:51.723 INFO Retransmitting MessageCounter:72266530 on exchange 18857i Send Cnt 3
[MatterTest] 01-12 18:28:52.049 INFO >>> [E:18857i S:42526 M:185962608 (Ack:72266530)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0001:07 (IM:WriteResponse)
[MatterTest] 01-12 18:28:52.076 INFO <<< [E:18857i S:42526 M:72266531 (Ack:185962608)] (S) Msg TX to 1:0000000012344321 [7609] [UDP:[fe80::8265:99ff:fec3:cf64%eth0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:52.076 INFO Step 14: Setting group key map on fabric 3
[MatterTest] 01-12 18:28:52.078 INFO >>> [E:18857i S:42526 M:185962609 (Ack:72266530)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:52.079 INFO Found an existing secure session to [5:0000000012344321]!
[MatterTest] 01-12 18:28:52.097 INFO <<< [E:18858i S:42528 M:40177813] (S) Msg TX to 5:0000000012344321 [94D6] [UDP:192.168.18.89%eth0:5540] --- Type 0001:06 (IM:WriteRequest)
[MatterTest] 01-12 18:28:52.099 INFO >>> [E:18857i S:42526 M:185962610 (Ack:72266530)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:52.124 INFO >>> [E:18857i S:42526 M:185962611 (Ack:72266530)] (S) Msg RX from 1:0000000012344321 [7609] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:52.785 INFO Retransmitting MessageCounter:40177813 on exchange 18858i Send Cnt 1
[MatterTest] 01-12 18:28:53.471 INFO Retransmitting MessageCounter:40177813 on exchange 18858i Send Cnt 2
[MatterTest] 01-12 18:28:54.408 INFO Retransmitting MessageCounter:40177813 on exchange 18858i Send Cnt 3
[MatterTest] 01-12 18:28:55.905 INFO Retransmitting MessageCounter:40177813 on exchange 18858i Send Cnt 4
[MatterTest] 01-12 18:28:58.566 ERROR Failed to Send CHIP MessageCounter:40177813 on exchange 18858i sendCount: 4 max retries: 4
[MatterTest] 01-12 18:28:59.551 INFO >>> [E:18858i S:42528 M:134177006 (Ack:40177813)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0001:07 (IM:WriteResponse)
[MatterTest] 01-12 18:28:59.638 INFO <<< [E:18858i S:42528 M:40177814 (Ack:134177006)] (S) Msg TX to 5:0000000012344321 [94D6] [UDP:192.168.18.89%eth0:5540] --- Type 0001:06 (IM:WriteRequest)
[MatterTest] 01-12 18:28:59.639 INFO >>> [E:18858i S:42528 M:134177007 (Ack:40177813)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:59.640 INFO >>> [E:18858i S:42528 M:134177008 (Ack:40177813)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:59.643 INFO >>> [E:18858i S:42528 M:134177009 (Ack:40177813)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:59.673 INFO >>> [E:18858i S:42528 M:134177010 (Ack:40177813)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:28:59.978 INFO Retransmitting MessageCounter:40177814 on exchange 18858i Send Cnt 1
[MatterTest] 01-12 18:29:00.355 INFO Retransmitting MessageCounter:40177814 on exchange 18858i Send Cnt 2
[MatterTest] 01-12 18:29:00.890 INFO Retransmitting MessageCounter:40177814 on exchange 18858i Send Cnt 3
[MatterTest] 01-12 18:29:01.749 INFO >>> [E:18858i S:42528 M:134177011 (Ack:40177814)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0001:07 (IM:WriteResponse)
[MatterTest] 01-12 18:29:01.777 INFO <<< [E:18858i S:42528 M:40177815 (Ack:134177011)] (S) Msg TX to 5:0000000012344321 [94D6] [UDP:192.168.18.89%eth0:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:29:01.778 INFO Step 14: Setting group key map on fabric 4
[MatterTest] 01-12 18:29:01.780 INFO Found an existing secure session to [8:0000000012344321]!
[MatterTest] 01-12 18:29:01.787 INFO >>> [E:18858i S:42528 M:134177012 (Ack:40177814)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:29:01.800 INFO <<< [E:18859i S:42529 M:213972027] (S) Msg TX to 8:0000000012344321 [5CA6] [UDP:192.168.18.89%eth0:5540] --- Type 0001:06 (IM:WriteRequest)
[MatterTest] 01-12 18:29:01.813 INFO >>> [E:18858i S:42528 M:134177013 (Ack:40177814)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:29:01.842 INFO >>> [E:18858i S:42528 M:134177014 (Ack:40177814)] (S) Msg RX from 5:0000000012344321 [94D6] --- Type 0000:10 (SecureChannel:StandaloneAck)
[MatterTest] 01-12 18:29:02.468 INFO Retransmitting MessageCounter:213972027 on exchange 18859i Send Cnt 1
[MatterTest] 01-12 18:29:03.100 INFO Retransmitting MessageCounter:213972027 on exchange 18859i Send Cnt 2
[MatterTest] 01-12 18:29:04.187 INFO Retransmitting MessageCounter:213972027 on exchange 18859i Send Cnt 3
[MatterTest] 01-12 18:29:05.819 INFO Retransmitting MessageCounter:213972027 on exchange 18859i Send Cnt 4
[MatterTest] 01-12 18:29:08.584 ERROR Failed to Send CHIP MessageCounter:213972027 on exchange 18859i sendCount: 4 max retries: 4
[MatterTest] 01-12 18:29:10.843 INFO SecureSession[0xffff8c0a07d0, LSID:42529]: State change 'kActive' --> 'kDefunct'
[MatterTest] 01-12 18:29:10.844 ERROR Time out! failed to receive write response from Exchange: 18859i
[MatterTest] 01-12 18:29:10.849 ERROR Exception occurred in test_TC_RR_1_1.
`
Environment

ESP-Matter Commit Id: matter release/1.3
ESP-IDF Commit Id: 5.2.1
SoC (eg: ESP32 or ESP32-C3): esp32-s3
Commissioner app and versions if present: certification tool Version: 2.10.2+spring2024
Device Logs (Please attach the log file):

[TC_RR_1.1_log_session_issue.txt](https://github.com/user-attachments/files/18390609/TC_RR_1.1_log_session_issue.txt)

bilalahmaddev, 2025-01-14:
can you please look into this as this is a certification blocker for us.

VaishaliAvhale, 2025-01-14:
Use the `--timeout 10000` parameter while executing this test case to resolve the timeout issue.

bilalahmaddev, 2025-01-14:
@VaishaliAvhale getting same issue with --timeout 10000 also attached are fresh logs
[TC_RR_1.1_log_session_issue.txt](https://github.com/user-attachments/files/18409000/TC_RR_1.1_log_session_issue.txt)

VaishaliAvhale, 2025-01-14:
There seems to be a small spelling mistake in the timeout parameter. Kindly correct it and try again.

bilalahmaddev, 2025-01-14:
Ah my bad, let me try again. Thanks

bilalahmaddev, 2025-01-14:
@VaishaliAvhale same issue please check
[TC_RR_1.1_log_session_issue.txt](https://github.com/user-attachments/files/18409478/TC_RR_1.1_log_session_issue.txt)

bilalahmaddev, 2025-01-14:
@VaishaliAvhale I reduced the number of endpoints to 3 and now test case is passing but our device has 12 endpoints. Is this something related to NVS size:
[TC_RR_1.1_log_session_issue.txt](https://github.com/user-attachments/files/18409792/TC_RR_1.1_log_session_issue.txt)

bilalahmaddev, 2025-01-14:
@VaishaliAvhale is this something related to this:
https://github.com/project-chip/connectedhomeip/issues/36882

bilalahmaddev, 2025-01-15:
@VaishaliAvhale can you please help me with this?

VaishaliAvhale, 2025-01-15:
Is this something related to NVS size?
>> Yes

bilalahmaddev, 2025-01-15:
@VaishaliAvhale I increased NVS but now getting this error on last step:
[image]

VaishaliAvhale, 2025-01-15:
We are looking into it and will get back to you soon. Thanks!

closed 2025-01-22 by bilalahmaddev