espressif/esp-matter #1241: Question about all_device_types_app example

tien1608, 2025-01-13:
I am trying all_device_types_app example. When I want to choose a light or fan device, how do I know which GPIO to connect the LED or fan to?

Jerry-ESP, 2025-01-14:
> I am trying all_device_types_app example. When I want to choose a light or fan device, how do I know which GPIO to connect the LED or fan to?

Hi @tien1608 ,
In the all_device_types_app example, there is only the driver of the fan, you can config the fan control GPIO in the menuconfig,  include CONFIG_STEP_MOTOR_A0_GPIO, CONFIG_STEP_MOTOR_A1_GPIO, CONFIG_STEP_MOTOR_B0_GPIO and CONFIG_STEP_MOTOR_B1_GPIO.

And there is no light driver in this example, if you want to control a actual light, you can refer to the light example. It can be set through this option: CONFIG_BSP_LED_RGB_GPIO, and the default light GPIO is the devkit's light GPIO, for esp32c3, it is GPIO8.

tien1608, 2025-01-14:
> > I am trying all_device_types_app example. When I want to choose a light or fan device, how do I know which GPIO to connect the LED or fan to?
> 
> Hi @tien1608 , In the all_device_types_app example, there is only the driver of the fan, you can config the fan control GPIO in the menuconfig, include CONFIG_STEP_MOTOR_A0_GPIO, CONFIG_STEP_MOTOR_A1_GPIO, CONFIG_STEP_MOTOR_B0_GPIO and CONFIG_STEP_MOTOR_B1_GPIO.
> 
> And there is no light driver in this example, if you want to control a actual light, you can refer to the light example. It can be set through this option: CONFIG_BSP_LED_RGB_GPIO, and the default light GPIO is the devkit's light GPIO, for esp32c3, it is GPIO8.
> 
> If you want to add the light driver in the all_device_types_app example, you can refer to the fan driver logic in the all_device_types_app/main/app_driver.cpp.

@Jerry-ESP Thank you, what motor or fan and driver can I use to illustrate this example?

closed 2025-01-20 by tien1608

InfiniteYuan, 2025-01-21:
Please take a look this doc：https://docs.espressif.com/projects/esp-iot-solution/en/latest/motor/bldc/bldc_overview.html