espressif/esp-matter #1242: Google Home: commissioning mutliple devices fails

mzakharocsc, 2025-01-13:
I am trying to commission 2 `light` matter example loaded boards with Google Home. First device commissioning always succeeds. While commissioning the second, commissioning fails and the first device gets removed from the Google Home as well. I am unable to have 2 devices connected to Google Home at the same time.

I have tried giving each device a unique descriminator,  while commissioning to Google Home, but this did not help. 

Commissioning and connecting both devices works fine with chip-tool and Home Assistant. 

Any insight on why this is happening with Google Home only is greatly appreciated.

Light example built using IDF 5.2.3 and  v1.4 branch of esp-matter-sdk a6da273b397275969741648be3e82577135192cb . Commisssioning using GHA version 3.27.1.6

jonsmirl, 2025-01-14:
You have to go into the Google Home console and enable the test device IDs for your account.

mzakharocsc, 2025-01-14:
I assume this is needed for every user (Google Account) that wants to commission the devices.  Is there a way to build a test device that is commissionable by any user account?

jonsmirl, 2025-01-14:
I don't believe there is a way to make the test ids work in a Google account other than by enabling them in the console.

jonsmirl, 2025-01-14:
You have to join CHIP and get a company ID assigned to make this work for all Google accounts.

I suspect this has to do with OTA updates. If products are using the same test IDs and someone sends an OTA update ... that OTA is going to go to all of those devices and it is not going to match some of them. Registering is used to sort that out.

mzakharocsc, 2025-01-14:
> I suspect this has to do with OTA updates.

I see the logic. Contrarily, how does one reconcile the fact that commissioning just one test device works while with multiple it does not. Surely, even with just one device commissioned, you would face the same OTA update problem.

jonsmirl, 2025-01-14:
I can commission any number of devices using the test ids on Google. I think you also have to register the device types? I set this up long ago an forgot how it works.

dhrishi, 2025-01-15:
@mzakharocsc I am assuming both the devices, being test/development devices, have the exact same onboarding payload (QR code). Please check the the page https://developers.home.google.com/matter/integration/pair - Warning, Pairing restrictions and Troubleshooting pairing sections should help.

mzakharocsc, 2025-01-15:
> Warning, Pairing restrictions and Troubleshooting pairing sections should help.

This warning is when trying to commissiong mutliple devices at the same time. I am commissioning one at a time, and have tried setting up multiple QR codes too by giving each device unique descriminator, no luck.

jonsmirl, 2025-01-15:
It doesn't matter if they all use the same QR code. I have a dozen devices commissioned on the same code. Make sure your have entered the right test VID/PID in the google console for the QR code you are using. 

I am also loading the corresponding test attestation certificates into my devices.  There is only an attestation certificate for FFF2/8001 so that is what I use for testing.

~~~
esp-matter-mfg-tool -cn "Test" -v 0xFFF2 -p 0x8001 --pai \
-k $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
-c $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
-cd $ESP_MATTER_PATH/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der \
--vendor-name "Lowpan" --product-name "Dimmer" --serial-num "12345678" \
--hw-ver-str "1.0.0" \
--passcode 20202021 --discriminator 3840
~~~

mzakharocsc, 2025-01-15:
Added fff2/8001 device to Google console. Removed the old 0xfff1/8000 device. waited 2.5 hours.  Flashed output of mfg-tool to factory partition. Performed commissioning using the new QR code.  Commissioning process displayed the new device logo set by the google console. However, there was no change in the outcome from the original posted behaviour.

jonsmirl, 2025-01-15:
Your Google Home device might be confused. Try factory resetting it. I had to do that several times to get old data out of mine.

mzakharocsc, 2025-01-16:
I found this question on Google support [forums](https://www.googlenestcommunity.com/t5/Smart-Home-Developer-Forum/How-many-test-Matter-devices-can-be-registered-on-Google-Home/m-p/645727#M7662):   Unfortunately, nothing I have tried has helped so far.

jonsmirl, 2025-01-16:
My Google Home device is enabled for beta testing. It is possible that it is running code you don't have. You can request to join testing, but it requires manual approval by Google.