espressif/esp-matter #1247: The software version reading has always been 1

MaplestoryAlen, 2025-01-17:
The software version I am currently reading has always been 1, and modifying the value of "CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION" has no effect. What I am still reading is 1. May I ask in which folder “CHIP_CONFIG_SOFTWARE_VERSION_NUMBER ” is defined? Can users freely change the software version

menuconfig:

CONFIG_ESP_SECURE_CERT_DS_PERIPHERAL=n
CONFIG_SEC_CERT_DAC_PROVIDER=y  
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"
CONFIG_FACTORY_DEVICE_INSTANCE_INFO_PROVIDER=y  

code：

/**
 * CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING
 *
 * A string identifying the software version running on the device.
 */
#ifndef CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING
#define CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION_STRING "1.0"
#endif

/**
 * CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION
 *
 * A monothonic number identifying the software version running on the device.
 */
#ifndef CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION
#define CHIP_DEVICE_CONFIG_DEVICE_SOFTWARE_VERSION 1
#endif

void CheckSoftwareVersion()
{
    uint32_t softwareVersion = 0;
    if (ConfigurationMgr().GetSoftwareVersion(softwareVersion) == CHIP_NO_ERROR)
    {
        printf("Current Software Version: %lu\n", static_cast<unsigned long>(softwareVersion));
    }
    else
    {
        printf("Failed to get Software Version.\n");
    }
}

File Path：esp-matter\connectedhomeip\connectedhomeip\src\platform\ESP32\ConfigurationManagerImpl.cpp

CHIP_ERROR ConfigurationManagerImpl::GetSoftwareVersion(uint32_t & softwareVer)
{
    softwareVer = CHIP_CONFIG_SOFTWARE_VERSION_NUMBER;
    return CHIP_NO_ERROR;
}

shubhamdp, 2025-01-17:
@MaplestoryAlen please take a look at this FAQ: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-12-firmware-version-number

MaplestoryAlen, 2025-01-20:
> [@MaplestoryAlen](https://github.com/MaplestoryAlen) please take a look at this FAQ: https://docs.espressif.com/projects/esp-matter/en/latest/esp32/faq.html#a1-12-firmware-version-number

> You can set project software version like this in Cmake file:
> 
> set(PROJECT_VER "v1.1") set(PROJECT_VER_NUMBER 1)

Thank you very much, the issue has been resolved

closed 2025-01-20 by MaplestoryAlen