espressif/esp-matter #1252: question about chip-tool

tien1608, 2025-01-20:
when I add a lamp device to chip-tool, I can control that device. But when I restarted the computer, chip-tool could not find the light device again. Why?

[log.txt](https://github.com/user-attachments/files/18477146/log.txt)

shubhamdp, 2025-01-20:
The CHIP Tool caches the configuration state in the `/tmp/` directory, and `/tmp` will be erased on reboot. To make the configuration persistent you can change the directory where chip-tool caches its configuration by using the command line option `--storage-directory`.

Please take a look at https://github.com/shubhamdp/connectedhomeip/blob/master/docs/development_controllers/chip-tool/chip_tool_guide.md#changing-storage-directory

tien1608, 2025-01-20:
> The CHIP Tool caches the configuration state in the `/tmp/` directory, and `/tmp` will be erased on reboot. To make the configuration persistent you can change the directory where chip-tool caches [...]

Thank you, my problem has been solved.

shubhamdp, 2025-01-20:
Please check the help content...
```
chip-tool pairing ble-wifi --help
.
.
.
[--storage-directory]:
  Directory to place chip-tool's storage files in.  Defaults to $TMPDIR, with fallback to /tmp
```

You will need to specify that every single time.

What's problem are you envisioning by not specifying it every time you use the command? If there is any, well you can do some scripting to append `--storage-directory <dir>` every time you use the command.

closed 2025-01-21 by tien1608