espressif/esp-matter #1253: How to decrypt esp_secure_cert.bin

abu-matterize, 2025-01-20:
Hello there,

The factory partition using `esp-matter-mfg-tool` with `--dac-in-secure-cert` flag, it generated `*_esp_secure_cert.bin` file. This file seems to be encrypted (is it?), if so, how can I decrypt this file?

shubhamdp, 2025-01-20:
@abu-matterize can you paste the complete command that you used for generating the partition? If you have used `-e` or `--encrypt` then the partition will be encrypted. Please check - https://github.com/espressif/esp-matter-tools/blob/main/mfg_tool/sources/mfg_tool.py#L541-L542

To dump the nvs partition, you can use`nvs_partition_tool` to dump the nvs partition.

```
python3 $IDF_PATH/components/nvs_flash/nvs_partition_tool/nvs_tool.py --help
```

To dump the content of the secure cert partition, you can use https://github.com/espressif/esp_secure_cert_mgr/tree/main/examples/esp_secure_cert_app to dump the contents.

abu-matterize, 2025-01-21:
Thanks for your response @shubhamdp. I've not used `-e` or `--encrypt` flag. I was wondering, if I use it, which key will be used to encrypt?

Here is the complete command: `esp-matter-mfg-tool --passcode 89674523 --discriminator 2245 -cd ../../TEST_CD_FFF1_8001.der -v 0xFFF1 --vendor-name Espressif -p 0x8001 --product-name Bulb --hw-ver 1 --hw-ver-str DevKit --pai -c ../../Esp-Development-PAI-Cert.pem --dac-cert ../../Esp-Development-DAC-01.pem --dac-key ../../Esp-Development-DAC-Key-01.pem --dac-in-secure-cert`

abu-matterize, 2025-01-21:
Tried [this](https://github.com/espressif/esp_secure_cert_mgr/tree/main/examples/esp_secure_cert_app) and below is the output.

<details>
<summary> Device Console Log [Click] </summary>

```sh
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
I (31) boot: ESP-IDF v5.3.2-dirty 2nd stage bootloader
I (31) boot: compile time Jan 21 2025 09:52:07
I (31) boot: Multicore bootloader
I (36) boot: chip revision: v1.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (53) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (77) boot:  1 factory          factory app      00 00 00020000 00100000
I (84) boot: End of partition table
I (89) esp_image: segment 0: paddr=00020020 vaddr=3f400020 size=0d09ch ( 53404) map
I (115) esp_image: segment 1: paddr=0002d0c4 vaddr=3ffb0000 size=0236ch (  9068) load
I (119) esp_image: segment 2: paddr=0002f438 vaddr=40080000 size=00be0h (  3040) load
I (122) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=28540h (165184) map
I (186) esp_image: segment 4: paddr=00058568 vaddr=40080be0 size=0bf58h ( 48984) load
I (212) boot: Loaded app from partition at offset 0x20000
I (212) boot: Disabling RNG early entropy source...
I (225) cpu_start: Multicore app
I (233) cpu_start: Pro cpu start user code
I (233) cpu_start: cpu freq: 160000000 Hz
I (233) app_init: Application information:
I (236) app_init: Project name:     esp_secure_cert_app
I (242) app_init: App version:      v2.2.0-77-gfe0ad6e
I (248) app_init: Compile time:     Jan 21 2025 09:52:03
I (254) app_init: ELF file SHA256:  8f91867c7...
I (259) app_init: ESP-IDF:          v5.3.2-dirty
I (264) efuse_init: Min chip rev:     v0.0
I (269) efuse_init: Max chip rev:     v3.99
I (274) efuse_init: Chip rev:         v1.0
I (279) heap_init: Initializing. RAM available for dynamic allocation:
I (286) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (292) heap_init: At 3FFB2C98 len 0002D368 (180 KiB): DRAM
I (298) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (305) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (311) heap_init: At 4008CB38 len 000134C8 (77 KiB): IRAM
I (319) spi_flash: detected chip: generic
I (322) spi_flash: flash io: dio
W (326) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (340) main_task: Started on CPU0
I (350) main_task: Calling app_main()
I (350) esp_secure_cert_app: Device Cert:
Length: 480
0l97s0
*H=0C10U

        ESP TEST PAI10
+|
  FFF110
+|
260119235959Z0C10U

                  ESP TEST PAI10
+|
  FFF110
+|
  80010Y0*H*H=B
I (370) esp_secure_cert_app: CA Cert:
Length: 452
004y}0
*H=0!10U
260119235959Z0C10Uelopment PAA0

                  ESP TEST PAI10
+|
  FFF110
+|
  80010Y0*H*H=B
I (390) esp_secure_cert_app: PEM KEY:
Length: 121
>!4.aoSވC
*H=DB
I (410) esp_secure_cert_app: Successfully parsed the certificate
I (410) esp_secure_cert_app: Successfully parsed the key
I (660) esp_secure_cert_app: Successfully signed the data
I (1140) esp_secure_cert_app: Successfully obtained and verified the contents of esp_secure_cert partition
I (1140) esp_secure_cert_app: Device Cert:
Length: 480
0l97s0
*H=0C10U

        ESP TEST PAI10
+|
  FFF110
+|
260119235959Z0C10U

                  ESP TEST PAI10
+|
  FFF110
+|
  80010Y0*H*H=B
I (1170) esp_secure_cert_app: Printing a list of TLV entries

I (1180) esp_secure_cert_tlv: TLV Entry No: 1
I (1180) esp_secure_cert_tlv: Type: 1, Subtype: 0, Data length: 480
I (1190) esp_secure_cert_tlv: Flags: 0

I (1190) esp_secure_cert_tlv: TLV Entry No: 2
I (1200) esp_secure_cert_tlv: Type: 0, Subtype: 0, Data length: 452
I (1200) esp_secure_cert_tlv: Flags: 0

I (1210) esp_secure_cert_tlv: TLV Entry No: 3
I (1210) esp_secure_cert_tlv: Type: 2, Subtype: 0, Data length: 121
I (1220) esp_secure_cert_tlv: Flags: 0
I (1220) main_task: Returned from app_main()
```
</details>

The certs seems to be okay, but how can get it to pem format? Is there some sort encoding happening in place while generating the `*_esp_secure_cert.bin`

shubhamdp, 2025-01-21:
@abu-matterize what you have dumped is the DER  encoded certificate, you can either encode it to PEM or hex dump that data and then may be `xxd` can help you with convert that hex dump to DER again and then use `openssl` commands to convert the cert from der to pem.

```
echo <continuous-hex-dump> | xxd -r -p > /tmp/cert.der
openssl x509 -in /tmp/cert.der -out /tmp/cert.pem -inform der -outform pem
```

abu-matterize, 2025-01-21:
@shubhamdp - Read back `esp_secure_cert` partition `esptool.py -p /dev/ttyUSB0 read_flash 0xd000 0x2000 cert.bin` and ran `md5sum cert.bin` to check if generated and dumped data are same and it is.

This means, I can work with `*_esp_secure_cert.bin` file, right?

Ran `openssl x509 -inform DER -in *_esp_secure_cert.bin -noout -text` 

```sh

Could not read certificate from cert.der
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
40D72D7BA37F0000:error:1E08010C:DECODER routines:OSSL_DECODER_from_bio:unsupported:../crypto/encode_decode/decoder_lib.c:101:No supported data to decode.  Input type: DER, Input structure: Certificate
Unable to load certificate
```

To double check ran `xxd` and `md5sum`-ed the output and the results are same.

`xxd -p 61b7e062-466c-4e7b-b060-706bde4ed570_esp_secure_cert.bin | xxd -r -p` produces

<details>
<summary> Possible DER?  [click] </summary>

```
^0l97s0
*H=0C10U

        ESP TEST PAI10
+|
  FFF110
+|
260119235959Z0C10U

                  ESP TEST PAI10
+|
  FFF110
+|
n:<Ejv9idyke{A9phbi1
                 UI`0^0
/0U#0i                 U00U0UrkG9t
      >eE_0
*H=H0E nIj?5!o3q8-KyI-^004y}0
*H=0!10U
260119235959Z0C10Uelopment PAA0

                  ESP TEST PAI10
+|
  FFF110
+|
  80010Y0*H*H=B
               e>5/w*Īt)`.BDŸG H9,e<cef0d000U0Ui
                                                >eE_0U#0        ^B0e20
*H=H0E!NGs@ٖ,,<
               ^
>!4.'t^y0w NaoSވC
n:<Ejv9idyke{A9ph
                 UI$
```
</details>

I've also attached the same .bin file here for your reference.

[esp_secure_cert_bin.tar.gz](https://github.com/user-attachments/files/18486116/esp_secure_cert_bin.tar.gz)

abu-matterize, 2025-01-21:
Let me give it a try.

abu-matterize, 2025-01-21:
Alright. hex dumping did the trick. Able to convert key and certs to `.pem`. If I chose `-e` in `esp-matter-mfg-tool` does it encrypt it with the DAC key? Or it uses different key?

If it uses different key, it needs to be in `nvs_keys` partition?

shubhamdp, 2025-01-21:
It does not use the DAC key to encrypt the partition it generates/uses the different one which is stored in nvs_key partition.

Please read more here: https://github.com/project-chip/connectedhomeip/blob/master/docs/platforms/esp32/flash_nvs_encryption.md

abu-matterize, 2025-01-21:
Thanks for helping.👍