espressif/esp-matter #1254: Linking CXX executable light.elf FAILED: light.elf

JesseDWatts, 2025-01-21:
I am having issues building the Light example. The first issue relates to a missing file related to bluetooth:

`ninja: error: '/home/jesse/esp/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/libble_app.a', needed by 'light.elf', missing and no known rule to make it`

To get around this I disabled BT. (Future issue to fix i thought)

The second issue is Linking CXX executable light.elf accompanied with some warnings regarding SEC_CERT_DAC_PROVIDER. 

OS: Ubuntu 22.04 on WSL
ESP-IDF Version: 5.5.0
Building for ESP32-C6

[idf_py_stdout_output_46922.txt](https://github.com/user-attachments/files/18487069/idf_py_stdout_output_46922.txt)
[idf_py_stderr_output_44462.txt](https://github.com/user-attachments/files/18487070/idf_py_stderr_output_44462.txt)

Thank you,

shripad621git, 2025-01-21:
@JesseDWatts ,both of the above issues will be resolved by the submodule update which is currently under process. The change will be available soon.
The below two links have the fixes for the above specified issues.
[esp32c6 ](https://github.com/project-chip/connectedhomeip/pull/36883) 
[esp_secure_cert](https://github.com/project-chip/connectedhomeip/pull/36985)

JesseDWatts, 2025-01-21:
@shripad621git thank you. Downgrading fixed those 2 issues. However, when I attempt to flash the firmware now I get:

`A fatal error occurred: Serial data stream stopped: Possible serial noise or corruption.
CMake Error at run_serial_tool.cmake:66 (message):

  /home/jesse/.espressif/python_env/idf5.2_py3.10_env/bin/python;;/home/jesse/esp/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32c6
  failed.`

Can you suggest anything before I try a fresh install?

shripad621git, 2025-01-22:
@JesseDWatts , can you please check the USB cable is properly connected. Possibly you can give it a try by using some other USB cable. For more reference, you can refer to the [common errors section ](https://docs.espressif.com/projects/esptool/en/latest/esp32c6/troubleshooting.html#common-errors) in esp-idf.

traderhut, 2025-01-27:
> As of now, we support the` esp-matter` with esp-idf version[ v5.2.3.](https://github.com/espressif/esp-matter?tab=readme-ov-file#supported-esp-idf-and-connectedhomeip-versions). And complete support for esp-idf `v5.5` will take quite some time as `esp-idf v5.5` is under development.

i pulled the default of the idf, and was able to build a program and load it on an ESP32 (blinking light) and that worked fine.

I've not been able to build any of the matter examples, getting what I assume is a linking error, although, the output is just returned error 1, with nothing above that that shows an error.  (It's been some 20 years since I had to work in C/C++, man it makes one appreciate c# when you see things like 'error 1')

In any event, do I need to roll back my entire 'idf' system to the v5.2.3 in order to build anything if I want it to support Matter?

traderhut, 2025-01-27:
> i pulled the default of the idf, and was able to build a program and load it on an ESP32 (blinking light) and that worked fine.
>
> I've not been able to build any of the matter examples, getting what [...]

I switched idf back to 5.2.3, and was able to get the blinky light again, but then when I went to load up the matter repo I got an error saying 'gn' was not installed, and I tried running the pip on the requirements.txt file (that was suggested somewhere in the help), and it said I was in an externally managed environment, and apparently Python has some cob up its * about having multiple packages and crud all installed globally... which begs the question why anyone would use Python when C# is out there.. but apparently, people actually enjoy these kind of nightmares... Anyway, I asked ChatGPT and it said, that the message that I got about creating a new environment was correct, and the best practice, so I did that... and re-ran the install.sh and it seemed to work... and now idf doesn't run anymore unless I use a different window, so apparently I have to reinstall it using this new environment which I created under the matter directory...  And ChatGPT agreed this is a royal mess and 'can get complicated when you have multiple things with different requirements', yeah, DLL heck they used to call it back in the day..

So... I presume the plan is to reinstall idf using this environment and maybe everything will be fine.. or perhaps, I just wipe out the entire thing and use a Raspberry Pi instead?... Been fighting this for over 2 hours now...  But at least, I got a little light blinking... no thanks to Matter code...  I was so sure I'd have it able to be turned on and off using Home Assistant...

traderhut, 2025-01-27:
I was monitoring my disk space as I was installing all this stuff.. and it appears to have sucked in about 30-50 GB of stuff in order to build a 1 MB application (there is only 2MB on the chip), which implies that it is sucking in the world in order to do a small (in terms of a windows application, a TINY) application.   Is all of this stuff really needed?   It is dying because someone is using 'gn' to do something and apparently that wasn't in the 50GB of stuff that it 'needed' to download in order to run.

I ran out of space multiple times loading this...

shripad621git, 2025-04-21:
@traderhut , you can try out the latest esp-matter code to build the esp-matter examples. The submodule update has the changes for c6-bt-lib path failure for idf versions. 
Please note the below points when working with newer esp-idf versions:
i) If you are changing the esp-idf version for esp-matter firmware, you need to checkout the esp-idf version you want to use.
for ex: If you want to switch to esp-idf v5.2.3 , `git checkout v5.2.3`
ii) Update the esp-idf submodules using `git submodule update --init --recursive -f.`
iii)Install the esp-idf using` ./install.sh.`
iv) export the esp-idf path using `. ./export.sh`

The esp-idf path needs to be exported each time you open a new firmware as mentioned [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#configuring-the-environment).

For esp-matter, if you are switching to latest version, please update the submodule in case of there is a submodule bump.
You can save the space in esp-matter by following the shallow clone approach as mentioned [here](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#configuring-the-environment).

dhrishi, 2025-05-15:
@JesseDWatts Please close this if your issue is resolved.

closed 2025-05-15 by JesseDWatts