espressif/esp-matter #1255: Window Covering Implementation help

kvanee, 2025-01-22:
**Describe the question/query that you have**
I need help updating a window covering attribute in reaction to a GPIO pin interrupt. 

**Additional context**
I have created a matter node, endpoint, window covering device, and cluster. 
I have setup an interrupt handler for the GPIO pin and verified via the monitor that the interrupt is called. 

In my interrupt handler I have the following code:
```
               chip::DeviceLayer::SystemLayer().ScheduleLambda([endpoint_id, position]() {
                    attribute_t * attribute = attribute::get(endpoint_id,
                                                            WindowCovering::Id,
                                                            WindowCovering::Attributes::CurrentPositionLiftPercentage::Id);

                    esp_matter_attr_val_t val = esp_matter_invalid(NULL);
                    attribute::get_val(attribute, &val);
                    val.val.u8 = static_cast<uint8_t>(position * 100);

                    attribute_t * attribute2 = attribute::get(endpoint_id,
                                                            WindowCovering::Id,
                                                            WindowCovering::Attributes::CurrentPositionLiftPercent100ths::Id);

                    esp_matter_attr_val_t val2 = esp_matter_invalid(NULL);
                    attribute::get_val(attribute2, &val2);
                    val2.val.u16 = static_cast<uint16_t>(position);

                    attribute::update(endpoint_id, WindowCovering::Id, WindowCovering::Attributes::CurrentPositionLiftPercentage::Id, &val);
                    attribute::update(endpoint_id, WindowCovering::Id, WindowCovering::Attributes::CurrentPositionLiftPercent100ths::Id, &val2);
                    ESP_LOGI("------INFO------", "Updated Attribute CurrentPositionLiftPercentage: %d", val.val.u8);
                    ESP_LOGI("------INFO------", "Updated Attribute CurrentPositionLiftPercentage100th: %d", val2.val.u16);
                });
```

And in the monitor output I see the following when toggling the pin: 

```
I (475192) ------INFO------: Updated Attribute CurrentPositionLiftPercentage: 100
I (475192) ------INFO------: Updated Attribute CurrentPositionLiftPercentage100th: 1
```

But I am not seeing any update for the device in my Home Assistant dashboard.

One other note, if I click to close the window cover from Home Assistant, I see the following in the monitor: 
`I (670182) chip[ZCL]: DownOrClose command received`
So I know the connection is working. 

But I also see this which I dont understand:
`WindowCovering has no delegate set for endpoint:1`
This is my first experience with Matter so please bear with me.

wqx6, 2025-01-24:
Hi, @kvanee when creating a window covering endpoint, you should set the delegate for the window covering cluster.
```
    window_covering_device::config_t window_covering_config;
    window_covering_config.window_covering.delegate = &wc_delegate;
    endpoint_t *ep = window_covering_device::create(node, &window_covering_config, ENDPOINT_FLAG_NONE, NULL);
```

So you should create an inheritance class of the [virtual class](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/window-covering-server/window-covering-delegate.h#L40) and override the two functions to handle the movement(`HandleMovement()`) or the stop motion(`HandleStopMotion()`).

If you look into the logic of [window covering cluster server implement](https://github.com/project-chip/connectedhomeip/blob/master/src/app/clusters/window-covering-server/window-covering-server.cpp), you would find the that the `HandleMovement()` will be called when the device receives commands of UpOrOpen, DownOrClose, GoToLiftValue, GoToLiftPercentage, GoToTiltValue, and GoToTiltPercentage. And you could find that the target values will be set before calling this function.So what you need to do in this function is to operate the motor and control the window covering move to the target value, and during the process, you should update the current value attributes.

And for the `HandleStopMotion()` function, you could find it is called when receiving StopMotion command, so you should stop the motor and update the current value attributes in this function.

As for the issue of that your window covering doesn't work well with HomeAssistant, I guess that the HomeAssistant controller shows the target value instead of the current values in the dashboard, so it doesn't update when you update the current values. And in my test with chip-tool, the subscription of current values works well.

kvanee, 2025-01-25:
I figured out my mistake. 

CurrentPositionLiftPercent100ths should be 10,000 when it's fully opened.

closed 2025-01-25 by kvanee