espressif/esp-matter #1257: Commissioning Problem on release/v1.4 branch. (Received ArmFailSafe (0s), CHIP Error 0x000000AC: Internal error)

halilakpinarr, 2025-01-23:
**Describe the bug**

We are doing matter bridge development with ESP32 Ethernet.
During commissioning with ESP Matter, we had no problem working with this commit ID. esp-matter commit id -> **7d3ec1ec981a77280d4b5c246a6ace4e57df4a47**

We migrated ESP Matter to release/v1.4 and now commissioning fails.
We migrated ESP-IDF from v5.2.1 to v5.2.3. 
ESP32 -> GeneralCommissioning: Received ArmFailSafe (0s)
chip-tool -> CHIP Error 0x000000AC: Internal error

**Environment**
- ESP-Matter Commit Id: release/v1.4
- ESP-IDF Commit Id: v5.2.3
- SoC (eg: ESP32 or ESP32-C3): esp32
- Device Logs (Please attach the log file):
[esp32_log.txt](https://github.com/user-attachments/files/18516009/esp32_log.txt)

- Host Machine OS: Ubuntu 22.04

- Home hub apps: apple tv, chip tool
- Commissioner's logs if present:
[chip-tool-log.txt](https://github.com/user-attachments/files/18516005/chip-tool-log.txt)

**ESP MFG tool**

esp-matter-mfg-tool --dac-in-secure-cert \
    -cn "Bridge" \
    -v 0xFFF2 \
    -p 0x8001 \
    -dm 2 \
    --discriminator 3840 \
    --vendor-name "manufacturer" \
    --product-name "test device" \
    --passcode 20202021 \
    --hw-ver-str 1.0 \
    --serial-num 12345678 \
    --product-label "bridge" \
    --pai \
    -k ~/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Key.pem \
    -c ~/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test/attestation/Chip-Test-PAI-FFF2-8001-Cert.pem \
    -cd ~/esp/esp-matter/connectedhomeip/connectedhomeip/credentials/test/certification-declaration/Chip-Test-CD-FFF2-8001.der

two binaries are flashed on the correct nvs partitions.

**SDK Config**
CONFIG_ENABLE_ESP32_FACTORY_DATA_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INSTANCE_INFO_PROVIDER=y
CONFIG_ENABLE_ESP32_DEVICE_INFO_PROVIDER=y
CONFIG_SEC_CERT_DAC_PROVIDER=y
CONFIG_ENABLE_SET_CERT_DECLARATION_API=y

CONFIG_CHIP_FACTORY_NAMESPACE_PARTITION_LABEL="fctry"

**Regarding Release/v1.4;**
I compiled the zigbee bridge application in release/v1.4, I changed only ethernet and telemetry parameters on it. 
Comissioning failed again, I think there is a problem with release/v1.4.

shubhamdp, 2025-01-23:
This is the error in your chip-tool logs, which says you used the wrong setup passcode.
```
[1737611961.000] [48645:48647] [SC] Failed to verify peer's MAC. This can happen when setup code is incorrect.
```

chip-tool command from your logs shows that you are using setup passcode as 7580858. But you built the partition with setup passcode 20202021 using mfg-tool.
```
chip-tool pairing onnetwork 0x1234 7580858
```

For chip-tool command help please add `--help` to the command, eg:
```
chip-tool pairing onnetwork --help
```

halilakpinarr, 2025-01-23:
I tried to enter the issue in Github.
I tried again with the default settings. I also get this error when I enter the installation password properly.
Actually this is exactly the error I am getting.

chip-tool pairing onnetwork 0x7283 20202021

[1737615380.691] [52234:52236] [EM] Failed to send Solitary ack for MessageCounter:85183945 on exchange 48783i:src/messaging/ExchangeContext.cpp:99: CHIP Error 0x00000002: Connection aborted
[1737615380.691] [52234:52236] [CTL] Commissioning complete for node ID 0x0000000000007283: src/app/ClusterStateCache.cpp:294: CHIP Error 0x000000CA: Interaction Model Error
[1737615380.691] [52234:52236] [TOO] Device commissioning Failure: src/app/ClusterStateCache.cpp:294: CHIP Error 0x000000CA: Interaction Model Error
[1737615380.691] [52234:52236] [DMG] ICR moving to [AwaitingDe]
[1737615380.691] [52234:52236] [IN] SecureSession[0x769070002be0]: Released - Type:1 LSID:6635
[1737615380.691] [52234:52234] [CTL] Shutting down the commissioner
[1737615380.691] [52234:52234] [CTL] Shutting down the controller
[1737615380.691] [52234:52234] [IN] Expiring all sessions for fabric 0x1!!
[1737615380.691] [52234:52234] [FP] Forgetting fabric 0x1
[1737615380.691] [52234:52234] [TS] Pending Last Known Good Time: 2023-10-14T01:16:48
[1737615380.691] [52234:52234] [TS] Previous Last Known Good Time: 2023-10-14T01:16:48
[1737615380.691] [52234:52234] [TS] Reverted Last Known Good Time to previous value
[1737615380.691] [52234:52234] [CTL] Shutting down the commissioner
[1737615380.691] [52234:52234] [CTL] Shutting down the controller
[1737615380.691] [52234:52234] [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1737615380.691] [52234:52234] [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1737615380.691] [52234:52234] [FP] Shutting down FabricTable
[1737615380.691] [52234:52234] [TS] Pending Last Known Good Time: 2023-10-14T01:16:48
[1737615380.691] [52234:52234] [TS] Previous Last Known Good Time: 2023-10-14T01:16:48
[1737615380.691] [52234:52234] [TS] Reverted Last Known Good Time to previous value
[1737615380.693] [52234:52234] [DL] Wrote settings to /tmp/chip_counters.ini
[1737615380.693] [52234:52234] [DL] NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1737615380.693] [52234:52234] [DL] Inet Layer shutdown
[1737615380.693] [52234:52234] [DL] BLE Layer shutdown
[1737615380.693] [52234:52234] [DL] System Layer shutdown
[1737615380.693] [52234:52234] [TOO] Run command failure: src/app/ClusterStateCache.cpp:294: CHIP Error 0x000000CA: Interaction Model Error

shubhamdp, 2025-01-23:
@halilakpinarr please please provide the device and chip-tool logs in the file, similar to what you have done in the issue description

halilakpinarr, 2025-01-23:
@shubhamdp logs are like this.

[chip-tool-log.txt](https://github.com/user-attachments/files/18516486/chip-tool-log.txt)
[esp32_log.txt](https://github.com/user-attachments/files/18516487/esp32_log.txt)

shubhamdp, 2025-01-23:
@halilakpinarr Thanks for the logs. Can you please pull the latest release/v1.4 and try with the below patch?

```
cd esp-matter
git pull origin release/v1.4
curl https://github.com/shubhamdp/esp-matter/commit/b1a4697eb7f2d2ae62f8623218cf7bd5b99de9dd.patch | git am
```

halilakpinarr, 2025-01-23:
@shubhamdp  

log level is "Debug", I did not update connectedhomeip, I need to make my own developments by downgrading to the old version.
Error log is below.

[esp32_log.txt](https://github.com/user-attachments/files/18522606/esp32_log.txt)

Commissioner;
Apple iphone 15 
IOS version: 18.1.1
Apple Home App version: Latest version of App Store

Another Topic,
Actually, we are planning to move to the application step for certification in a short time, can you suggest me a stable version? We can also stay with a slightly older version rather than staying with versions that increase product risk.

bzbarsky-apple, 2025-01-23:
Looking at the logs attached in https://github.com/espressif/esp-matter/issues/1257#issuecomment-2609288268, ConnectMaxTime has nothing to do with anything: chip-tool is sending an AttestationRequest command and the server is responding with FAILURE, which is why commissioning fails.

halilakpinarr, 2025-01-24:
@shubhamdp 
There is a topic that I think may help you.

I downgraded and was working on this release.
a6da273b397275969741648be3e82577135192cb

I flashed the application code compiled with this version (a6da273b397275969741648be3e82577135192cb)
Then I created qr code in esp-matter release/v1.4 and flashed it.
The commissioning in a6da273b397275969741648be3e82577135192cb failed again.
Then, i create a new qr code in a6da273b397275969741648be3e82577135192cb, and flashed again. commissioning completed successfully.
I believe that this problem may occur after a development related to mfg_tool.
Maybe it will help.

shubhamdp, 2025-01-27:
By default, the factory data provider implementation reads the Certification Declaration (CD) from the 'chip-factory'. You have enabled `CONFIG_ENABLE_SET_CERT_DECLARATION_API=y` option, which enables the API which lets you set the CD from the application and the configured CD will be used for during attestation validation.

Please check the documentation here, https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32SecureCertDACProvider.h#L35-L55.

Either you can se the CD using this API, or please disable that option so that the certification declaration that you have provided during `mfg-tool` will be used.

shubhamdp, 2025-01-27:
In https://github.com/espressif/esp-matter/issues/1257#issuecomment-2612288607, commits that you have mentioned in are all same(https://github.com/espressif/esp-matter/commit/a6da273b397275969741648be3e82577135192cb).

For setting the CD through application, we will push an example code for reference.

shubhamdp, 2025-01-27:
@halilakpinarr Patch that I provided you in https://github.com/espressif/esp-matter/issues/1257#issuecomment-2609169247 has been merged into release/v1.4 in 30af618a6e962623a0098ad6a33b468f33dc49c7, please pull the latest changes when testing.

shubhamdp, 2025-02-02:
@halilakpinarr did above comments help resolve the problem?

halilakpinarr, 2025-02-03:
@shubhamdp  I have not had a chance to try it yet, but I will let you know when I do.

dhrishi, 2025-03-03:
@halilakpinarr I am closing this now. Please re-open if you face issues, when you test.

closed 2025-03-03 by dhrishi