espressif/esp-matter #1258: HTTP Server initialization fails after Matter Started

bilalahmaddev, 2025-01-24:
HTTP Server initialization fails after Matter Started and it works fine if Matter is not started and device is not commissioned. Can't we use HTTP server with Matter? 

httpd_server_init: error in listen (112)

[image]

jonsmirl, 2025-01-24:
I run HTTPD with Matter, make sure you aren't out of memory. My devices all use PSRAM

bilalahmaddev, 2025-01-24:
@jonsmirl SPIRAM is enabled. Is there any specific CONFIG to enable HTTPD to use SPIRAM for memory allocation?

bilalahmaddev, 2025-01-25:
This is resolved after reducing FreeRTOS tasks stack memory allocation.

closed 2025-01-25 by bilalahmaddev

jonsmirl, 2025-01-25:
Move your stacks into PSRAM.
~~~
    StackType_t* mem_task_stack = (StackType_t*)heap_caps_calloc(1, 4096, MALLOC_CAP_SPIRAM);
    static StaticTask_t mem_task;
    xTaskCreateStatic(memTask, "mem", 4096, NULL, 0, mem_task_stack, &mem_task);
~~~

bilalahmaddev, 2025-01-26:
@jonsmirl both these options should be enabled for this:
SPIRAM_USE_CAPS_ALLOC and SPIRAM_USE_MALLOC ?

jonsmirl, 2025-01-26:
move your other stacks to PSRAM