espressif/esp-matter #1260: How can I edit the fan featuremap for the ESP32-C6?

adailydose, 2025-01-27:
I would like to adjust the featuremap for the fan endpoint on the ESP32-C6 matter implementation. However, I'm not sure where to adjust the featuremap.

jadhavrohit924, 2025-01-29:
`FeatureMap` attribute can only be updated if you have added particular feature on the cluster. The [document](https://docs.espressif.com/projects/esp-matter/en/latest/esp32/developing.html#features) says how to add features to a cluster, you can do similarly for the fan.

dhrishi, 2025-02-10:
@adailydose Were you able to update the feature map as per your requirements? Please close the issue if that was done.

dhrishi, 2025-03-03:
The above instructions should help you update the features/feature map. Please re-open if you face issues

closed 2025-03-03 by dhrishi