espressif/esp-matter #1261: How to properly combine sensor and icd_app example?

dmartauz, 2025-01-27:
I have successfully tested both sensor example (adapted for SHT41) and also icd_app example.
Now I would like to implement regular measuring of temperature and humidity into icd_app example.
In original sensor example measuring process is triggered by a timer. However I think in Matter sensor ICD it makes sense to synchronise measuring process with switch from Idle mode to Active mode.

Is there any callback which can be used for this purpose?

dmartauz, 2025-01-29:
@xbrunosousa could you please share your code? I tried using timer, however I2C operations fail if called from timer callback while in idle mode (light sleep).

Jerry-ESP, 2025-02-07:
It is an I2C retention issue, has been fixed in idf 5.3 and later version, please use the correct idf version, e.g. v5.3.2

dmartauz, 2025-02-08:
Tested with v5.3.2 successfully, thank you.

closed 2025-02-08 by dmartauz