espressif/esp-matter #1262: Recommissioning Fails After First Successful Commissioning – Device Not Found Until Network is Manually Changed

Lenin-Weinkling, 2025-01-29:
After commissioning my device successfully using the commissioning window, I attempted to change the network credentials by recommissioning the device from the Google Home app. However, the commissioning process gets stuck at "Searching for device" until I manually switch my network back to the previously commissioned one.

**Steps to Reproduce:**

1) Press the button to open the commissioning window using the function below:

void OpenCommissioningWindow()
{
    PlatformMgr().ScheduleWork([](intptr_t arg) {
        uint16_t timeout = static_cast<uint16_t>(arg);

        CHIP_ERROR err = Server::GetInstance()
                             .GetCommissioningWindowManager()
                             .OpenBasicCommissioningWindow(System::Clock::Seconds32(timeout));

        if (err == CHIP_NO_ERROR) {
            ChipLogProgress(AppServer, "Commissioning window opened successfully for %d seconds.", timeout);
        } else {
            ChipLogError(AppServer, "Failed to open commissioning window: %s", ErrorStr(err));
            matter_pairing = false;
        }
    }, static_cast<intptr_t>(300));
}

2) Commission the device successfully to a Wi-Fi network.
3) Try to change the network credentials by recommissioning the device using the Google Home app.
4) Observe that the commissioning process gets stuck on "Searching for device."
5) The device only becomes discoverable again if I manually switch my network back to the previously commissioned one.

wqx6, 2025-03-07:
You cannot change the network that the end-device connects by recommissioning as the BLE will not advertise commissionable service after the first commissioning. The only way to change the network is:
1. Send ArmFailSafe command of GeneralCommissioning cluster to the end-device
2. Send RemoveNetwork command of NetworkCommissioning cluster to the end-device
3. Send AddOrUpdateWifiNetwork command of NetworkCommissioning cluster to the end-device with the new network info
4. Send ConnectNetwork command of NetworkCommissioning cluster to the end-device so that the device will connect to the new network.
5. Send CommissionComplete command of GeneralCommissioning cluster to the end-device

For more details, you can refer the the NetworkCommissioning Test Case [TC-CNET-4.11] in MatterTestPlans document.

dhrishi, 2025-03-11:
@Lenin-Weinkling Please check if your question is answered

Lenin-Weinkling, 2025-03-11:
Thank you for your detailed response! I’m working on developing a Matter product and using apps like Google Home, Alexa, and HomeKit for commissioning. I was able to re-enable BLE commissioning by setting **CONFIG_USE_BLE_ONLY_FOR_COMMISSIONING=n,** and now the BLE advertising for commissioning shows up even after the first commissioning. However, I’m still facing an issue where the device setup fails during connection to the new network during recommissioning. It seems the only way to successfully recommission to a new network is by performing a factory reset on the esp-matter.

Is there any way to solve this without requiring a factory reset? I’d like to allow users to change network credentials seamlessly through the apps without losing other settings. Any guidance would be greatly appreciated!