espressif/esp-matter #1263: Unpairing device callbacks

pavel808, 2025-01-29:
I am using the following API to decommission / unpair  a  Matter end device via my application : 
`DeviceCommissioner::UnpairDevice(NodeId remoteDeviceId)`

Is there currently some way to register a user-defined callback to get notified when the unpairing procedure finishes, with a success or fail? Thanks.

jadhavrohit924, 2025-02-05:
If you want to remove the current fabric, you should use the `Server::GetInstance().GetFabricTable().Delete()` API instead. Then, you will get the `kFabricRemoved` event in the applications.

pavel808, 2025-02-07:
Hi @jadhavrohit924 I saw your reply to this question to use : `Server::GetInstance().GetFabricTable().Delete() ` , but the reply seems to be removed?

This call would be only for the end device side right? 

What I am trying to do is de-commission devices from my Matter Controller application and have an application callback invoked on the Matter application side when it is completed stating a success or fail. Thanks.

closed 2025-02-26 by dhrishi