espressif/esp-matter #1264: Error while compiling light example with esp_matter v1.3 with esp-idf v5.1.5 tag for esp32c6

milindmovasha, 2025-01-30:
**Describe the bug**

I am trying to compile the esp_mater light example using esp-idf from the tag https://github.com/espressif/esp-idf/releases/tag/v5.1.5 and esp_matter v1.3 for esp32c6.

I get the following error:
ninja: error: '/home/milindm/projects/jack/sdk/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/libble_app.a', needed by 'light.elf', missing and no known rule to make it
ninja failed with exit code 1,

The library exists at different path.
/home/milindm/projects/jack/sdk/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6/libble_app.a

Used the following steps:

-   Setup `esp-idf`

    ```
    git clone --recursive https://github.com/espressif/esp-idf.git
    cd esp-idf
    git checkout v5.1.5
    git submodule update --init --recursive
    ./install.sh
    . ./export.sh
    cd ..
    ```

-   Setup `esp-matter`

    ```
    git clone -b release/v1.3 --depth 1 https://github.com/espressif/esp-matter.git
    cd esp-matter
    git submodule update --init --depth 1
    ./connectedhomeip/connectedhomeip/scripts/checkout_submodules.py --platform esp32 --shallow
    ./install.sh
    pip install -r requirements.txt
    . ./export.sh
    cd ..
    ```
- Compilation
cd esp-matter/examples/light
idf.py set-target esp32c6
idf.py build

Any idea how to resolve this?
**Environment**
- ESP-Matter Commit Id: release/v1.3 ba10f4942b753699cd898e5f17ce0af8384cce44
- ESP-IDF Commit Id: v5.1.5 454782bb6b214beb8a23273312ca4d833c504810
- SoC: esp32c6
- Device Logs (Please attach the log file): Please refer to description

- Host Machine OS: Linux
- Host Machine Python version: Python 3.11.10

- Commissioner app and versions if present: NA
- Home hub app and versions if present: NA
- Commissioner's logs if present: NA

**Any additional details**
None

shripad621git, 2025-01-30:
@milindmovasha , we recommend using the `idf v5.2.1` with esp-matter `release/v1.3 `as mentioned [here](https://github.com/espressif/esp-matter/tree/release/v1.3?tab=readme-ov-file#supported-esp-idf-and-connectedhomeip-versions). For `esp-idf v5.1.5`, you can fix the issue by specifying the  correct path [here](https://github.com/espressif/connectedhomeip/blob/593d5c6f63a62e017e4ced43183049f2805a9db8/config/esp32/components/chip/CMakeLists.txt#L424) which for esp-idf 5.1.5 is `esp/esp-idf/components/bt/controller/lib_esp32c6/esp32c6-bt-lib/esp32c6/libbleapp.a`.

shripad621git, 2025-02-07:
@milindmovasha , is the issue resolved?

milindmovasha, 2025-02-16:
> [@milindmovasha](https://github.com/milindmovasha) , we recommend using the `idf v5.2.1` with esp-matter `release/v1.3 `as mentioned [...]

@shripad621git With the above change I am able to compile with idf v5.1.5 successfully.

milindmovasha, 2025-02-16:
Changed the line in the file esp-matter/connectedhomeip/connectedhomeip/config/esp32/components/chip/CMakeLists.txt. as follows:
```
list(APPEND chip_libraries "${bt_dir}/controller/lib_${target_name}/${target_name}-bt-lib/${target_name}/libble_app.a"
```

dhrishi, 2025-03-03:
Closing this now

closed 2025-03-03 by dhrishi