espressif/esp-matter #1266: on-off light endpoint is not working good

Teorozzo, 2025-02-03:
Hi everyone, I'm trying to set-up and example of on_off light but I don't know if the patch (finder here (https://github.com/espressif/esp-matter/issues/241) still works after all the updates.
I don't even know if the matter example light is good for what I am doing.
I set up the example in an ESP32 with 1 red led and one button, if I use the button I can switch correctly the led, instead, using the ESP32-S3 where I run the controller example correctly paired, I can just turn on the light using "OnOff on..." but I cannot turn it off, neither using "OnOff off...", nor OnOff toggle....".

jadhavrohit924, 2025-02-04:
Hi @Teorozzo, you don’t need the patch anymore. With ESP32 you'll have to set the `CONFIG_BSP_LEDS_NUM` as it is 0 by default. After setting the flag in the `menuconfig` ( `Component config → Board Support Package (generic) → LEDs ->Number of LEDs in BSP` )  the example works fine on esp32.

Teorozzo, 2025-02-05:
Maybe I'm wrong but setting the number of led to zero, I'm not able to turn on/off an external led using GPIO.
I'm trying to do that connecting the led at G2 pin and control it from and external matter controller which I made

jadhavrohit924, 2025-02-05:
You should set `CONFIG_BSP_LEDS_NUM` to the number of LEDs you have.

dhrishi, 2025-02-11:
@Teorozzo Did the suggestion help?

Teorozzo, 2025-02-13:
Unfortunately, the LED can be switched on via the Matter controller, but it never actually turns off. I have a single LED on GPIO2 and a button on GPIO4. The issue is that when I attempt to turn the LED off through the Matter controller, the brightness gets set to 1 rather than 0, so the LED stays lit.

jadhavrohit924, 2025-02-19:
@Teorozzo Sorry for the delayed response, I got your issue. As per matter spec, when the `OnOff` and `LevelControl` clusters are on the same endpoint and LevelControl has enabled the `OnOff feature`, they have interdependencies. If you just remove the `OnOff feature` from your `LevelControl` cluster the LED will turn `OFF`.

Suppose you have an OnOff feature on the LevelControl cluster with an OnOff server on the same endpoint. In that case, the `ON` means the `maximum level` of brightness that your LevelControl cluster has defined, and `OFF` means the `minimum level` of brightness. Since the MinLevel attribute is set to 1 in the example as it is the minimum range the attribute can have as per spec, when you try to turn off the light it sets the brightness to zero. This is an expected behaviour.

Teorozzo, 2025-02-26:
Thanks @jadhavrohit924, you've been crystal clear! Could you tell me how to remove the LevelControl from the cluster?

closed 2025-03-04 by Teorozzo