espressif/esp-matter #1267: Can we increase NVS Partition Size in esp-matter code

Ajay2803, 2025-02-04:
Hello

I'm working with Light Example and using an ESP32-C6-MINI-1. 4MB is the total size of the flash memory.
I have one file in my Project Directory **Partition.csv** Inside that i saw that my NVS Flash Partition is only 48kb, and WiFi credentials are occupying around 8kb. so only 40kb is Remaining. , and i want to store some more data in NVS, more that 40kb so is it possible to increase the NVS Partition size and if yes how can i increase it, is there any kind of document or thread you can share, which will be very helpful for us.

shripad621git, 2025-02-04:
@Ajay2803 , you can go through this [partition-table document](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/partition-tables.html) for the partition-table related information.

jonsmirl, 2025-02-05:
just make the partition larger, there are no restrictions.

closed 2025-02-06 by Ajay2803