espressif/esp-matter #1268: Deep Sleep for generic_switch demo

brobinson88, 2025-02-04:
Newbie question! Is it possible to use deep sleep with the generic switch demo on an esp32-c6 or is light sleep the only option? Lets say for example I have two GPIO buttons on GPIO pins 6 and 7, I want the device to wake up from deep sleep, perform the buttons task and go back to into deep sleep.

Jerry-ESP, 2025-02-07:
For the matter device, deep sleep is not possible, when device enter deep sleep mode, it almost equal to power off, the device will be offline and need to rebuild case session after wake up.

brobinson88, 2025-02-18:
Just thought I would share with anyone wanting to know: yes you can create a generic switch as a sleepy end device that uses deep sleep!

I modified the icd_app demo to create a generic switch endpoint by copying most of the code from the generic_switch demo. Works perfectly! Typical consumption in deep sleep is around 0.06 - 0.08mA on an esp32-c6 using matter over wifi.  the switch is also much more responsive than I was expecting! I thought there might be a large delay from the gpio wake up but it's super fast!

closed 2025-02-18 by brobinson88