espressif/esp-matter #1269: TimeSynchronizationCluster: Wrong epoch time conversion ?

caipiblack, 2025-02-05:
**Describe the bug**
When we set the UTC time of a node running on esp32, the time being set seems to be wrong (or wrongly displayed).

For example:

```
I (2248785) chip[TS]: Last Known Good Time: 2025-02-05T17:51:33
I (2248788) chip[TS]: New proposed Last Known Good Time: 2025-02-05T17:51:33
I (2248796) chip[TS]: Updating Last Known Good Time to 2025-02-05T17:51:33
I (2248804) chip[DL]: Real time clock set to 1738777893 (0125/01/05 17:51:33 UTC)
```

**Environment**
FireBeetle 2 ESP32-C6

**Any additional details**
1. Use any simple example
2. Configure the example to implement the TimeSynchronization cluster:

```

static esp_err_t esp_matter_command_callback_set_utc_time(const ConcreteCommandPath &command_path, TLVReader &tlv_data,
                                                      void *opaque_ptr)
{
    chip::app::Clusters::TimeSynchronization::Commands::SetUTCTime::DecodableType command_data;
    CHIP_ERROR error = chip::app::DataModel::Decode(tlv_data, command_data);
    if (error == CHIP_NO_ERROR) {
        emberAfTimeSynchronizationClusterSetUTCTimeCallback((chip::app::CommandHandler *)opaque_ptr, command_path, command_data);
    }
    return ESP_OK;
}

void main(void)
{
....
endpoint_t *root_endpoint = endpoint::get(node,0);
esp_matter::cluster::time_synchronization::config_t time_config;
static chip::app::Clusters::TimeSynchronization::DefaultTimeSyncDelegate time_sync_delegate;
time_config.delegate = &time_sync_delegate;
esp_matter::cluster_t *time_cluster = esp_matter::cluster::time_synchronization::create(root_endpoint, &time_config, CLUSTER_FLAG_SERVER | CLUSTER_FLAG_CLIENT);  
ABORT_APP_ON_FAILURE(time_cluster != nullptr, ESP_LOGE(TAG, "Failed to create time_sync_cluster"));
esp_matter::command::create(time_cluster, chip::app::Clusters::TimeSynchronization::Commands::SetUTCTime::Id, COMMAND_FLAG_ACCEPTED, esp_matter_command_callback_set_utc_time);
...
}
```
3. Commission the device using chiptool
4. Send a command "SetUTCTime" to the device:
```
# WARNING: The UTC time is passed as microseconds, but it must be the difference between NOW and 1st January 2000
# Be carefull here, it's not unix timestamp
chip-tool timesynchronization set-utctime 792093093930488 4 1 0
```

Here a simple tool to get the compatible epoch time:

```
#!/bin/bash

now=$(date +%s%6N)
epoch_2000=$(date -d "2000-01-01 00:00:00" +%s%6N)
diff=$((now - epoch_2000))
echo "$diff"
```

It's possible that the problem is due to a wrong conversion, but I am not sure. As the Matter epoch time is based on 2000 year, (not like unix timestamp)

caipiblack, 2025-02-05:
It seems that only the prinf it wrong, because I arrive to read the correct datetime using this:

```
static void vDisplayDateTime(void) {
    chip::System::Clock::Microseconds64 cTMs;
    chip::System::SystemClock().GetClock_RealTime(cTMs);
    time_t unixEpoch = std::chrono::duration_cast<chip::System::Clock::Seconds32>(cTMs).count();
	char buf[50];
    tm calendarTime{};
    localtime_r(&unixEpoch, &calendarTime);
	printf("The date and time is %s", asctime_r(&calendarTime, buf));
}
```

So at this point, this issue is very minor.

pimpalemahesh, 2025-03-03:
@caipiblack The issue has been fixed, and the fix will be available after the submodule update. Could you please close this issue?

caipiblack, 2025-03-03:
Thanks it’s all good

closed 2025-03-03 by caipiblack