espressif/esp-matter #1271: BLE Integration in Matter example code

Ajay2803, 2025-02-08:
Hi,
I am working on Matter device Development and I want to use BLE after commissioning, so i took reference from this thread (https://github.com/espressif/esp-matter/issues/1207), In this i found one patch which includes BLE integration in Matter. I have used BLE for advertising means in peripheral mode and i am able to successfully compile and build this code.

But when I enables BLE after commissioning I am not able connect it by using my Phone or Laptop it shows this kind of error:
[image]
and when i tried to compile example which is used for this integration ~esp/esp-idf/examples/bluetooth/nimble/bleprph then it is working properly and this is it output:

[image]

Can you help me to solve this issue so i can go with further modification.

Ajay2803, 2025-02-08:
I am facing the same issue as the last guy mentioned in this thread https://github.com/espressif/esp-matter/issues/906

shripad621git, 2025-02-13:
@Ajay2803, can you please apply the below patch and give it a try. 

[ble_bonding.txt](https://github.com/user-attachments/files/18784257/ble_bonding.txt)

Ajay2803, 2025-02-14:
Thank-you for your helpful reply.
After using your patch i am able to pair my device with my Mobile phone's Bluetooth. I know that matter is using custom profile and service for commissioning over BLE, Now as i said i used this thread (https://github.com/espressif/esp-matter/issues/1207) as a reference and in this, user is just Reseting GATT service using ble_gatts_reset(); due to this every thing get erase, Now i want to control an HID Device let say TV, So how can i set HID profile and HID service in this source?

If we applying custom HID Profile and service, then does it breaking any matter condition or may effect in certification issue?
and if you can tell us drawback of this then it will also helpful for us.