espressif/esp-matter #1272: ESP32H2 SuperMini Matter over Thread Example

kemalicecek, 2025-02-09:
Hi, I am trying to develop a thread accessory for myself. I found an  esp32h2 module from [TEMU](https://share.temu.com/8JxwBOtWP3A), price was good and since I don't want wifi and this is a quite small board, I wanted to take my chances. Now my problem is, with my C6 I believe I can use just thread and add it to my Apple Homekit but with h2 I did it few times but now I can't pass commissioning with default settings. 

I am using IDF-5.2.3 and IDF-Matter release/v1.4

[c6_logs.txt](https://github.com/user-attachments/files/18722994/c6_logs.txt)

[h2_logs.txt](https://github.com/user-attachments/files/18722992/h2_logs.txt)

Jerry-ESP, 2025-02-10:
For thread device, you should have a Homepod in your Apple Home as a thread border router. For C6, it is a matter over WiFi device in your log, not a thread device.

dhrishi, 2025-03-03:
@kemalicecek As mentioned above, please configure your ESP32-C6 as a Matter Thread device. It should work fine. If you need pre-built firmware, you can download from here: https://espressif.github.io/esp-launchpad/?flashConfigURL=https://espressif.github.io/esp-matter/launchpad.toml (Use thread_matter_light and choose ESP32C6)

I am closing this issue now. Please re-open in case of any problems

closed 2025-03-03 by dhrishi

adyanth, 2025-08-07:
Where is the source code for the thread_matter_light? I have been going in circles in the esp-matter repo, unable to find it :(