espressif/esp-matter #1274: Issues encountered during authentication [TC-RR-1.1],[TC-DA-1.2]

MaplestoryAlen, 2025-02-11:
Question 1： [TC-RR-1.1] Details=Failed Step 11: MaxGroupsPerFabric < 4 * counted_groups_clusters, Extras=None

[image]

    I have modified the program definition CHIP_CONFIG-MAX-GROUPS-PER-FABRIC，change from 1 to 2.
 The response given by the laboratory test is that the DUT has two endpoints with groups, and MaxGroupsPerFabric needs to be greater than 8. 
 However, I am not sure if this change is possible because I do not know how to verify this result

 #ifndef CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC
  #define CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC  2 //1 - SDK default
 #endif
#ifndef CHIP_CONFIG_MAX_GROUPS_PER_FABRIC
#define CHIP_CONFIG_MAX_GROUPS_PER_FABRIC (4 * CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC)
#endif

Question 2：[TC-DA-1.2] Details=0 not found in [1, 2] Certification type is out of range, Extras=None

[image]
 
The reason for the test failure is that the TYPE of the CD certificate should not be 0，
My method of generating a CD is as follows：

 ./chip-cert gen-cd -K ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem \
    -C ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem \
    -O esp_dac_1575_2024.der -f 1 \
    -V 0x1575 -p 0x2024 -d 0x0016 -c "CSA00000SWC00000-01" -l 0 -i 0 -n 1 -t 0

How to modify parameters so that the TYPE of the CD certificate is not 0？

Jerry-ESP, 2025-02-11:
> Question 1： [TC-RR-1.1] Details=Failed Step 11: MaxGroupsPerFabric < 4 * counted_groups_clusters, Extras=None
>
> [image]
>
> ```
> I have modified the program definition [...]
 
```
Yes, you just need change the CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC equal to the endpoints count with group cluster. Just in this file:   esp-matter/connectedhomeip/connectedhomeip/src/lib/core/CHIPConfig.h
```

> Question 2：[TC-DA-1.2] Details=0 not found in [1, 2] Certification type is out of range, Extras=None
>
> [image]
>
> The reason for the test failure is that the TYPE of the CD certificate should not be 0， [...]

```
You should change the param --certification-type to 1 to pass the certification in the CD generated command.
```

MaplestoryAlen, 2025-02-11:
> ```
> Yes, you just need change the CHIP_CONFIG_MAX_GROUP_ENDPOINTS_PER_FABRIC equal to the endpoints count with group cluster. Just in this file: [...]

--certification-type 

[image]

./chip-cert gen-cd -K ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem -C ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Cert.pem -O esp_dac_1575_2024.der -f 1 -V 0x1575 -p 0x2024 -d 0x0016 -c "CSA00000SWC00000-01" -l 0 -i 0 -n 1 -t 1

Is that how to change the parameter '- t' from 0 to 1 or 2?

Jerry-ESP, 2025-02-11:
> --certification-type
>
> [image]
>
> ./chip-cert gen-cd -K ../../credentials/test/certification-declaration/Chip-Test-CD-Signing-Key.pem -C [...]

Correct!

MaplestoryAlen, 2025-02-11:
> Correct!

Thank you very much for your answer

closed 2025-02-11 by MaplestoryAlen