espressif/esp-matter #1275: TC-IDM-10.4 failing TriggerEffect command

bilalahmaddev, 2025-02-11:
TC-IDM-10.4 failing TriggerEffect because we are using TriggerEffect command, how to disable it? 
problem: An element found on the device, but the corresponding PICS I.S.C40.Rsp was not found in pics list
[

[TC-IDM-10-4_2025_02_11_12_07_30.log](https://github.com/user-attachments/files/18746369/TC-IDM-10-4_2025_02_11_12_07_30.log)

](url)

Environment

ESP-Matter Commit Id: matter release/1.3
ESP-IDF Commit Id: 5.2.1
SoC (eg: ESP32 or ESP32-C3): esp32=s3

VaishaliAvhale, 2025-02-12:
Hi @bilalahmaddev, could you please let us know which device type you are using while testing this test case?

bilalahmaddev, 2025-02-12:
@VaishaliAvhale Hi, I am using OnOff Plugin and dimmable plugin device types

VaishaliAvhale, 2025-02-12:
Hi @bilalahmaddev,

For OnOff Plug-in Unit and Dimmable Plug-in Unit device types, the Identify Cluster and its TriggerEffect command are mandatory, so you can't remove it.

You can fix TC-IDM-10.4 by updating the TriggerEffect command of the Identify Cluster to true in PICS.

bilalahmaddev, 2025-02-12:
@VaishaliAvhale As per specification TriggerEffect command of Identify cluster are not mandatory

VaishaliAvhale, 2025-02-12:
Please check the element requirements for the OnOff Plug-in Unit and Dimmable Plug-in Unit device types in the device library specification. The TriggerEffect command is optional for the Identify Cluster, but it is mandatory for both device types, so it has been added in both firmwares.

bilalahmaddev, 2025-02-12:
How to disable it TriggerEffect command for the Identify Cluster only? otherwise it fails TC-IDM-10.4.

VaishaliAvhale, 2025-02-12:
@bilalahmaddev, As the TriggerEffect command is mandatory for both device types, we can't disable it. Please refer to the Matter v1.3 Device Library Specifications:

Section 5.1.5 (Element Requirements) for On/Off Plug-in Unit device requirements.
Section 5.2.5 (Element Requirements) for Dimmable Plug-in Unit device requirements.

bilalahmaddev, 2025-02-16:
Thanks for the confirmation.

closed 2025-02-16 by bilalahmaddev