espressif/esp-matter #1277: request for an example on reading attribute from device to device without binding

albrownwood, 2025-02-12:
Hello,
I am trying to read a temperature attribute on a sensor device from another device in the same fabric without establishing a binding. The fabricID and nodeId are known values. I looked at the esp_matter::client::connect and esp_matter_client but couldn't wrap my head around how to use these. Is there an existing snippet example I could start from?

wqx6, 2025-02-12:
You can start with the light switch example which sends subscription command to the light example, https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/app_main.cpp#L104-L109. If you want to use read instead of subscribe, please replace the `client::interaction::subscribe::send_request()` with `client::interaction::read::send_request()` at https://github.com/espressif/esp-matter/blob/main/examples/light_switch/main/app_driver.cpp#L82

albrownwood, 2025-02-16:
Thanks for pointing this to me @wqx6, looks like a good starting point

albrownwood, 2025-02-17:
Thanks @wqx6 I have been able to send a request for reading the Thermostat::LocalTemperature on another device.

In the 'onAttributeData' callback I get a Null pointer for the data and on the target device I get a 'Access denied' message in the log, the ecosystem is Apple Home. I was hoping that things like temperature values are by default configured as readable by other devices on the same Fabric, was that a wrong assumption?
Is there a way to configure the devices in that way?

wqx6, 2025-03-03:
> In the 'onAttributeData' callback I get a Null pointer for the data and on the target device I get a 'Access denied' message in the log, the ecosystem is Apple Home. I was hoping that things like [...]

I am afraid of that the Apple Ecosystem will not allow a device from a third party vendor to be able to read attributes from another device in their fabric, The Read interaction requires View privilege but the reader cannot be added to the ACL of the server. So it will get an `Access denied` error.

> Is there a way to configure the devices in that way?

The only way is to add the two devices to your fabric instead of Ecosystem's fabric and modify ACL of the server.

albrownwood, 2025-03-08:
makes sense, thanks @wqx6 closing this one.

closed 2025-03-08 by albrownwood