espressif/esp-matter #1278: User defined mDNS does not work

MaplestoryAlen, 2025-02-12:
In the Matter SDK program, I first connected the device to WIFI and then tried to obtain relevant information through custom mDNS broadcasting. However, the current situation only shows the default mDNS of the SDK, and user-defined mDNS is invalid, mdns_delegate_hostname_add(delegated_hostname, &addr4); The error returned is: E (8499) mDNS: Error in mdns_delegate_hostname_add: ESP_ERR_INVALID_STATE
   

static void app_event_cb(const ChipDeviceEvent *event, intptr_t arg)
{
    case chip::DeviceLayer::DeviceEventType::kInterfaceIpAddressChanged:

	       if(event->InterfaceIpAddressChanged.Type == chip::DeviceLayer::InterfaceIpChangeType::kIpV4_Assigned)
	       {

	        }
		 else if(event->InterfaceIpAddressChanged.Type == chip::DeviceLayer::InterfaceIpChangeType::kIpV4_Lost)
		 {

		 }
		 else if(event->InterfaceIpAddressChanged.Type == chip::DeviceLayer::InterfaceIpChangeType::kIpV6_Assigned)
		 {
			 app_add_mdns_srvc();
		 }	
                break;
}

void app_add_mdns_srvc(void)
{
	
    int nodeId = 0x123456;
    esp_err_t esp_err = ESP_FAIL;

    esp_netif_t *intf = esp_netif_get_default_netif();
    
    char *delegated_hostname;
    if (-1 == asprintf(&delegated_hostname, "Xz-mDns-%d", nodeId)) {
        ESP_LOGE("mDNS", "Memory allocation failed for delegated_hostname");
        return;
    }

    mdns_ip_addr_t addr4, addr6;
    addr4.addr.type = ESP_IPADDR_TYPE_V4;
    esp_netif_ip_info_t info;
    esp_netif_get_ip_info(intf, &info);
    addr4.addr.u_addr.ip4 = info.ip;
    
    addr6.addr.type = ESP_IPADDR_TYPE_V6;
    esp_netif_get_ip6_linklocal(intf, &addr6.addr.u_addr.ip6);
    addr4.next = &addr6;
    addr6.next = NULL;

    esp_err = mdns_delegate_hostname_add(delegated_hostname, &addr4);
    if (esp_err != ESP_OK) 
	{
        ESP_LOGE("mDNS", "Error in mdns_delegate_hostname_add: %s", esp_err_to_name(esp_err));
        free(delegated_hostname);
        return;
    }
    esp_err = mdns_service_add_for_host(delegated_hostname, "_http", "_tcp", delegated_hostname, 80, NULL, 0);
    if (esp_err != ESP_OK) 
	{
        ESP_LOGE("mDNS", "Error in mdns_service_add_for_host: %s", esp_err_to_name(esp_err));
        free(delegated_hostname);
        return;
    }

    ESP_LOGI("mDNS", "mDNS service added successfully");
    
 
    free(delegated_hostname);
}

log:

[image]

[image]

wqx6, 2025-02-12:
Looks like you are using minimal mdns and platform mdns at the same time, if you want to add your mdns service. You should disable minimal mdns in menuconfig
```
CONFIG_USE_MINIMAL_MDNS=n
```
Also you should make sure that the `mdns_delegate_hostname_add()` is called after `mdns_init()`.

MaplestoryAlen, 2025-02-13:
> Looks like you are using minimal mdns and platform mdns at the same time, if you want to add your mdns service. You should disable minimal mdns in menuconfig
> 
> ```
> CONFIG_USE_MINIMAL_MDNS=n
> ```
> 
> Also you should make sure that the `mdns_delegate_hostname_add()` is called after `mdns_init()`.

I want to add DN and DT values from TXT in the broadcast. How do I add them? SDK is not available by default

[image]

Similar to the following：
[image]

MaplestoryAlen, 2025-02-17:
> > I want to add DN and DT values from TXT in the broadcast. How do I add them?
> 
> To enable DT mdns txt record, you should enable `ENABLE_COMMISSIONABLE_DEVICE_TYPE` in menuconfig To enable DN mdns txt record, you can add the following line to the platform configuration file `connectedhomeip/src/platform/ESP32/CHIPDevicePlatformConfig.h` or your project configuration file.
> 
> ```
> #define CHIP_DEVICE_CONFIG_ENABLE_COMMISSIONABLE_DEVICE_NAME 1
> #define CHIP_DEVICE_CONFIG_DEVICE_NAME "YourDeviceName"
> ```

The problem has been resolved, thank you very much

closed 2025-02-17 by MaplestoryAlen