espressif/esp-matter #1279: Error occurred after configuring 'CONFIG_USE_MINIMAL_MDNS=n'

MaplestoryAlen, 2025-02-17:
Matter SDK version: 1.2

 Error occurred after configuring 'CONFIG_USE_MINIMAL_MDNS=n',I can read mDNS broadcasts, but it seems that the broadcasts are not functioning properly，I don't know why this is happening, but there are no such issues when the default setting is set toCONFIG_USE_MINIMAL_MDNS=y

[image]

[image]

[image]

log:
  W (21) boot: Unicore bootloader
I (969) cpu_start: Unicore app
W (969) flash HPM: HPM mode is optional feature that depends on flash model. Read Docs First!
W (970) flash HPM: HPM mode with DC adjustment is disabled. Some flash models may not be supported. Read Docs First!
W (982) flash HPM: High performance mode of this flash model hasn't been supported.
I (990) MSPI Timing: Flash timing tuning index: 2
I (995) octal_psram: vendor id    : 0x0d (AP)
I (1000) octal_psram: dev id       : 0x02 (generation 3)
I (1006) octal_psram: density      : 0x03 (64 Mbit)
I (1012) octal_psram: good-die     : 0x01 (Pass)
I (1017) octal_psram: Latency      : 0x01 (Fixed)
I (1023) octal_psram: VCC          : 0x01 (3V)
I (1028) octal_psram: SRF          : 0x01 (Fast Refresh)
I (1034) octal_psram: BurstType    : 0x01 (Hybrid Wrap)
I (1040) octal_psram: BurstLen     : 0x01 (32 Byte)
I (1045) octal_psram: Readlatency  : 0x02 (10 cycles@Fixed)
I (1051) octal_psram: DriveStrength: 0x00 (1/1)
I (1064) MSPI Timing: PSRAM timing tuning index: 2
I (1064) esp_psram: Found 8MB PSRAM device
I (1067) esp_psram: Speed: 120MHz
I (1174) mmu_psram: Instructions copied and mapped to SPIRAM
I (1361) mmu_psram: Read only data copied and mapped to SPIRAM
I (1361) cpu_start: Single core mode
I (1372) cpu_start: Pro cpu start user code
I (1372) cpu_start: cpu freq: 160000000 Hz
I (1372) cpu_start: Application information:
I (1375) cpu_start: Project name:     light
I (1380) cpu_start: App version:      1.0
I (1385) cpu_start: Compile time:     Feb 17 2025 09:29:55
I (1391) cpu_start: ELF file SHA256:  f22843e1807f211e...
I (1397) cpu_start: ESP-IDF:          v5.2.1
I (1402) cpu_start: Min chip rev:     v0.0
I (1407) cpu_start: Max chip rev:     v0.99 
I (1412) cpu_start: Chip rev:         v0.2
I (1416) heap_init: Initializing. RAM available for dynamic allocation:
I (1424) heap_init: At 3FCB8948 len 00030DC8 (195 KiB): RAM
I (1430) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (1436) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (1442) heap_init: At 600FE010 len 00001FD8 (7 KiB): RTCRAM
I (1449) esp_psram: Adding pool of 3456K of PSRAM memory to heap allocator
I (1456) spi_flash: detected chip: gd
I (1460) spi_flash: flash io: dio
W (1465) i2c: This driver is an old driver, please migrate your application code to adapt `driver/i2c_master.h`
I (1476) sleep: Configure to isolate all GPIO pins in sleep state
I (1482) sleep: Enable automatic switching of GPIO sleep configuration
I (1489) coexist: coex firmware version: 77cd7f8
I (1495) coexist: coexist rom version e7ae62f
I (1500) main_task: Started on CPU0
I (1504) main_task: Calling app_main()
I (5462) app_main: thermostat created with endpoint_id 1
I (5463) app_main: fan created with endpoint_id 2
I (5463) app_main: fan created with endpoint_id 3
I (5480) pp: pp rom version: e7ae62f
I (5480) net80211: net80211 rom version: e7ae62f
I (5483) wifi:wifi driver task: 3fccf8d0, prio:23, stack:6656, core=0
I (5497) wifi:wifi firmware version: a9f5b59
I (5498) wifi:wifi certification version: v7.0
I (5498) wifi:config NVS flash: enabled
I (5500) wifi:config nano formating: disabled
[... 169 lines cut ...]
I (6950) chip[DL]: Device already advertising, stop active advertisement and restart
I (6958) NimBLE: GAP procedure initiated: stop advertising.

I (6965) NimBLE: GAP procedure initiated: advertise; 
I (6970) NimBLE: disc_mode=2
I (6974) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (6984) NimBLE: 

I (6987) app_main:  Interface WiFi Connectivity Change...

I (7145) main_task: Returned from app_main()
I (7502) esp_netif_handlers: sta ip: 192.168.50.80, mask: 255.255.255.0, gw: 192.168.50.1
I (7502) chip[DL]: IP_EVENT_STA_GOT_IP
I (7504) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.50.80/255.255.255.0 gateway 192.168.50.1
I (7516) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (7521) app_main: Interface  Internet Connectivity Change...

I (7528) app_main: Interface IP Address changed
I (7533) app_main: kIpV4_Assignedd

I (7585) chip[DIS]: Updating services using commissioning mode 1
E (7586) mdns: mdns_service_remove_for_host(6508): Service doesn't exist
I (7589) chip[DIS]: Advertise commission parameter vendorID=5493 productID=8228 discriminator=0790/03 cm=1

[sdkconfig.txt](https://github.com/user-attachments/files/18818100/sdkconfig.txt)

MaplestoryAlen, 2025-02-26:
Is there anyone who can provide guidance or reply

MaplestoryAlen, 2025-02-26:
> 您使用的是哪个版本的 mdns？您可以在 中找到它。`your_example/managed_components/espressif__mdns/idf_component.yml`

dependencies:
  idf:
    version: '>=5.0'
description: Multicast UDP service used to provide local network service and host
  discovery.
documentation: https://docs.espressif.com/projects/esp-protocols/mdns/docs/latest/en/index.html
issues: https://github.com/espressif/esp-protocols/issues
repository: git://github.com/espressif/esp-protocols.git
repository_info:
  commit_sha: b65cff3a0b7c666501884c9eae4416616ce26493
  path: components/mdns
url: https://github.com/espressif/esp-protocols/tree/master/components/mdns
version: 1.4.2

wqx6, 2025-02-26:
The error logs of `Invalid state or arguments` or `Service doesn't exist` is OK for the esp-matter version that you are using and they can be ignored, as the check of mdns service is not added for your esp-matter. https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32DnssdImpl.cpp#L181. But it will not affect the functionality of mdns advertisement.

[image]
And I can find there is a right mdns record in your avahi-browse results, isn't it right?

MaplestoryAlen, 2025-02-26:
> The error logs of `Invalid state or arguments` or `Service doesn't exist` is OK for the esp-matter version that you are using and they can be ignored, as the check of mdns service is not added for [...]

Yes, mDNS is already working and you can see the broadcast. I also encountered the same problem in the matter SDK V1.4 version, where the logs reported errors. Can I ignore this issue?

V1.4 SDK log:

[image]

V1.4 SDK mDNS idf_component.yml:

dependencies:
  idf:
    version: '>=5.0'
description: Multicast UDP service used to provide local network service and host
  discovery.
documentation: https://docs.espressif.com/projects/esp-protocols/mdns/docs/latest/en/index.html
issues: https://github.com/espressif/esp-protocols/issues
repository: git://github.com/espressif/esp-protocols.git
repository_info:
  commit_sha: dea5f1c4316edde20fb316612934a950e65a0e77
  path: components/mdns
url: https://github.com/espressif/esp-protocols/tree/master/components/mdns
version: 1.7.0

wqx6, 2025-02-26:
> Can I ignore this issue?

You can ignore the error log. But could you please let us know where does `mdns_service_remove_for_host` get called? I think that we have added the check for all the codes that call `mdns_service_remove_for_host()`. https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32DnssdImpl.cpp#L181-L184 https://github.com/project-chip/connectedhomeip/blob/master/src/platform/ESP32/ESP32DnssdImpl.cpp#L214-L225

MaplestoryAlen, 2025-02-26:
> You can ignore the error log. But could you please let us know where does `mdns_service_remove_for_host` get called? I think that we have added the check for all the codes that call [...]

In the V1.4 version of the program I just created, I did not call any mDNS operations. I just configured it in menuconfig as "CONFIG_USE_MINIMAL_MDNS=n”,and then an error log appeared ->mdns: mdns _service remove for host(6675): service doesn't exist

sdkconfig:

[v1.4_sdkconfig.txt](https://github.com/user-attachments/files/18979294/v1.4_sdkconfig.txt)

wqx6, 2025-02-26:
Could you add a line of `printf("line %d\n", __LINE__);` before `mdns _service_remove_for_host()` and `mdns _service_remove()` in `ESP32DnssdImpl.cpp` so that we can know which service removing function is called?

MaplestoryAlen, 2025-02-27:
> Could you add a line of `printf("line %d\n", __LINE__);` before `mdns _service_remove_for_host()` and `mdns _service_remove()` in `ESP32DnssdImpl.cpp` so that we can know which service removing function is called?

Program Add Location：

    // Remove service before adding it
    if (mdns_service_exists(service->mType, GetProtocolString(service->mProtocol), nullptr))
    {
        mdns_service_remove_for_host(service->mName, service->mType, GetProtocolString(service->mProtocol), nullptr);
		printf("line %d\n", __LINE__);
    }

CHIP_ERROR EspDnssdRemoveServices()
{
    if (mdns_service_exists("_matter", "_tcp", nullptr))
    {
        mdns_service_remove("_matter", "_tcp");
		printf("line %d\n", __LINE__);
    }
    if (mdns_service_exists("_matterc", "_udp", nullptr))
    {
        mdns_service_remove("_matterc", "_udp");
		printf("line %d\n", __LINE__);
    }
    if (mdns_service_exists("_matterd", "_udp", nullptr))
    {
        mdns_service_remove("_matterd", "_udp");
		printf("line %d\n", __LINE__);
    }
    return CHIP_NO_ERROR;
}

log:

[image]

I (1800) wifi:new:<10,0>, old:<1,0>, ap:<255,255>, sta:<10,0>, prof:1
I (1804) wifi:state: init -> auth (b0)
I (1811) wifi:state: auth -> assoc (0)
I (1818) wifi:state: assoc -> run (10)
I (1836) wifi:connected with Gerlsair_XizhenElec, aid = 19, channel 10, BW20, bssid = 08:bf:b8:65:4b:b8
I (1836) wifi:security: WPA2-PSK, phy: bgn, rssi: -35
I (1849) mdns_mem: mDNS task will be created from internal RAM
I (1850) wifi:pm start, type: 1

I (1852) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1858) wifi:set rx beacon pti, rx_bcn_pti: 14, bcn_timeout: 25000, mt_pti: 14, mt_time: 10000
I (1866) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (1867) esp_matter_cluster: Cluster plugin init common callback
I (1881) esp_matter_core: Dynamic endpoint 0 added
I (1883) esp_matter_core: Dynamic endpoint 1 added
I (1894) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000002 is 0 **********
I (1900) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000002 is 0 **********
I (1911) esp_matter_attribute: ********** W : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000002 is 1 **********
I (1923) esp_matter_attribute: ********** R : Endpoint 0x0000's Cluster 0x0000002A's Attribute 0x00000003 is null **********
I (1941) IOT_WIFI: Wifi connected
I (1945) wifi:<ba-add>idx:0 (ifx:0, 08:bf:b8:65:4b:b8), tid:6, ssn:1, winSize:64
I (1949) bridge: Endpoint ID: 33619970
I (1951) bridge: Device Type ID: 114
I (1956) esp_matter_bridge: Cluster plugin init for the new added endpoint
I (1963) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000003's Attribute 0x00000001 is 5 **********
I (1974) esp_matter_attribute: ********** R : Endpoint 0x0002's Cluster 0x00000006's Attribute 0x0000FFFC is 2 **********
I (1986) esp_matter_core: Dynamic endpoint 2 added
I (1993) bridge: Endpoint ID: 33619971
I (1995) bridge: Device Type ID: 114
I (2001) esp_matter_bridge: Cluster plugin init for the new added endpoint
I (2008) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000003's Attribute 0x00000001 is 5 **********
I (2019) esp_matter_attribute: ********** R : Endpoint 0x0003's Cluster 0x00000006's Attribute 0x0000FFFC is 2 **********
I (2031) esp_matter_core: Dynamic endpoint 3 added
I (2036) espnow: esp-now Version: 2.5.1
I (2041) espnow: device already connected to SSID: Gerlsair_XizhenElec
I (2048) ESPNOW: espnow [version: 1.0] init
I (2052) espnow: mac: 30:30:f9:27:bc:6c, version: 2
I (2057) espnow: Enable main task
I (2061) espnow: main task entry
I (2062) app_espnow: espnow init success.
I (2071) app_espnow: espnow init completed.
I (2075) main_task: Returned from app_main()
I (2940) esp_netif_handlers: sta ip: 192.168.50.141, mask: 255.255.255.0, gw: 192.168.50.1
I (2968) app_main: Interface IP Address changed
I (2970) app_main: kIpV4_Assignedd
I (2969) wifi:<ba-add>idx:1 (ifx:0, 08:bf:b8:65:4b:b8), tid:0, ssn:1, winSize:64
E (2983) mdns: mdns_service_remove_for_host(6675): Service doesn't exist
line 184
I (3080) IOT_WIFI: Connected with IPv6 Address:fe80:0000:0000:0000:3230:f9ff:fe27:bc6c
I (3080) IOT_WIFI: Got ipv6-fe80:0000:0000:0000:3230:f9ff:fe27:bc6c
I (3086) app_main: Interface IP Address changed
I (3090) app_main: kIpV6_Assigned
line 218
line 184
E (3098) mdns: mdns_service_remove_for_host(6675): Service doesn't exist
line 184
I (6075) ACInterface: read iot ac number 16
I (6075) espnow_bridge: espnow switch found: 1a:2b:3c:4a:5b:00, initiator type: 258
I (6077) espnow_bridge: Bridged node for 1a:2b:3c:4a:5b:00 ESP-NOW device on endpoint 2 has been created
I (6087) ACInterface: read addr number 0
I (6292) espnow_bridge: espnow switch found: 1a:2b:3c:4a:5b:01, initiator type: 258
I (6292) espnow_bridge: Bridged node for 1a:2b:3c:4a:5b:01 ESP-NOW device on endpoint 3 has been created
I (6299) ACInterface: read addr number 1
I (6559) ROUTE_HOOK: Received RIO
I (6559) ROUTE_HOOK: prefix FD1F:F965:BBF7:1:: lifetime 1800
I (8085) ROUTE_HOOK: Hook already installed on netif, skip...
I (8091) app_main: Interface IP Address changed
I (8096) app_main: kIpV6_Assigned
line 218
line 184
E (8103) mdns: mdns_service_remove_for_host(6675): Service doesn't exist
line 184

wqx6, 2025-04-18:
I think I know the reason for this issue. The mdns_service_remove() will post the remove-service process queue to the mdns task, so it will not remove the service immediately. And when the `EspDnssdRemoveServices()` is called multiple times in short interval, the remove-service process queue will be post multiple times, so there will be error logs. But this error log is OK and you can ignore them.

MaplestoryAlen, 2025-04-18:
Okay, thank you very much for your answer

closed 2025-04-18 by MaplestoryAlen