espressif/esp-matter #1280: Set a Custom IPK in esp_matter_controller_client

crivi30, 2025-02-17:
Hello everyone!

I need to use a custom IPK instead of the default one in the `matter_controller_client` after calling `setup_commissioner()`. However, the current implementation sets the **default** IPK inside `setup_commissioner()`, and there is no direct way to override it with a custom value.

## Steps to Reproduce

1. Call `setup_commissioner()` from `matter_controller_client`.
2. Observe that the default IPK is set using:
   ```cpp
   chip::ByteSpan default_ipk = chip::GroupTesting::DefaultIpkValue::GetDefaultIpk();
   ESP_RETURN_ON_FALSE(chip::Credentials::SetSingleIpkEpochKey(group_data_provider, fabric_index, default_ipk,
                                                                compressed_fabric_id_span) == CHIP_NO_ERROR,
                        ESP_FAIL, TAG, "Failed to set ipk for commissioner fabric"); 

There is no mechanism to pass a custom IPK instead of the default one.

Is there any existing way to set a custom IPK after calling `setup_commissioner()`?
If not, would modifying `setup_commissioner()` to accept a custom IPK be the best approach?

Thanks  in advance

wqx6, 2025-02-21:
I think you can call the `chip::Credentials::SetSingleIpkEpochKey()` with your custom IPK after `setup_commissioner()`. It will overwrite the existing default IPK.

crivi30, 2025-02-21:
`chip::Credentials::SetSingleIpkEpochKey` requires `m_group_data_provider`, which is private in `matter_controller_client` class. Would there be another way to access or set it?

dhrishi, 2025-03-03:
@crivi30 Please close this if your question is answered

closed 2025-03-03 by crivi30