espressif/esp-matter #1283: RGB Light indication in multiple_on_off_plugin_units example

Ajay2803, 2025-02-19:
Hi,

I am using multiple_on_off_plugin_units example for reference in my development, and i want to use RGB led Indication in that, Can anyone how can i use RGB Indication, i am unable to find out any thread or example related to this, if anyone has any source or example related to this please help me.

I have tried to do manage RGB manually but when it connects to the network while commissioning, it conflicts over there, it is unable to find out the actual state of relay at that time, i relay is ON then LED indications shows it OFF and if it is OFF LED indication shows it ON. Only when it's connects to the network.

shubhamdp, 2025-02-25:
What do you want the RGB indicator to do? Whats expected behaviour of that indicator? What sort of example code are you looking for? For messing with RGB peripheral you can find the code in `examples/light/main/app_driver.cpp`, it has sample code snippets and functions for on-off/level/color control.

Are you saying that during commissioning when device attempts to connect to WiFi network, you are unable to get the correct state of the relay?

> i relay is ON then LED indications shows it OFF and if it is OFF LED indication shows it ON. Only when it's connects to the network.

I think your driver code is not setting the LED correctly, can you check if relay state is being actually reflected/set on the LED correctly? Does it work when not connected to the network?

Also, can you elaborate the use case, what you want to do or expecting by adding an RGB LED. What you want to do when it connects to network?

shubhamdp, 2025-02-26:
@Ajay2803 Exact things that you need for your use case may not be present. What I can tell you is when you can perform actions on the LEDs.

For WiFi... you can [register the event handler](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32/api-reference/system/esp_event.html#_CPPv426esp_event_handler_register16esp_event_base_t7int32_t19esp_event_handler_tPv) and check for `WIFI_EVENT_STA_CONNECTED` and `WIFI_EVENT_STA_DISCONNECTED` events and perform the required action. An example implementation can be found in [esp-idf/examples/wifi/getting-started/station](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c#L73)

I hope this would work for wifi events (If you'd like to use IP events to perform these indications you can use those events `...GOT_IP` AND `LOST_IP`).

For control (BLUE LED), you can use use the [app_attribute_update_cb](https://github.com/espressif/esp-matter/blob/main/examples/light/main/app_main.cpp#L150). When someone toggles your relay from the Matter protocol this will be called, and you can update your blue LED. If there are any more consumers of this, you should consider synchronizing the state with them as well... (just an eg: relay being trigged through manufacturer app and this will be caught in different callback).

On bootup, you can set the wifi led to red and once connected I assume you turn on the green one and turn off the red one. For blue, you may need to check the relay's state and update this accordingly.

You can check the esp-matter's light examples for driver implementation design.

closed 2025-03-10 by Ajay2803