espressif/esp-matter #1284: ESP-Skainet + ESP-Matter

Teorozzo, 2025-02-19:
Does anyone used the Skainet library and sent the command via matter using di ESP-Matter library?

shubhamdp, 2025-02-25:
I'm not sure if it uses the skainet internally but there's this example in esp-box repository which is matter-switch and can be toggled using `hi esp` wake word https://github.com/espressif/esp-box/tree/master/examples/matter_switch#control-devices

What is your use case? What are you building an end device or controller/commissioner?

Teorozzo, 2025-02-26:
Thank you, @shubhamdp! The example might be exactly what I was looking for. My idea is to set up a Matter controller using the Korvo-S3 V2, where I can use speech recognition to receive commands and control end devices.

shubhamdp, 2025-02-26:
@Teorozzo glad to help. If your question is answered can be please close the issue.

closed 2025-02-26 by Teorozzo