espressif/esp-matter #1286: ESP32-H2 cannot be compiled on the Matter Light use case

aaa147147, 2025-02-21:
[image]

May I ask, is this because ESP32-H2 is not supported?

wqx6, 2025-02-21:
Hi, could you please provide IDF version, esp-matter commit, and the complete building log?

aaa147147, 2025-02-24:
@wqx6 Thank you for your reply. The information is as follows

IDF version: v5.4.0
esp-matter commit: 
```
aaa147147@DESKTOP-5UFAS5T:~/ESP-IDF-Tool/esp-matter$ git log --oneline -1
e2a83f8 (HEAD -> main, origin/main, origin/HEAD) Merge branch 'zigbee_bridge_update' into 'main'
```
complete building log

[ESP-IDF-building log.log](https://github.com/user-attachments/files/18935873/ESP-IDF-building.log.log)

wqx6, 2025-02-24:
Sorry for that there is an issue which enables Border Agent by default for Thread end device on the IDF that you are using. please cherry-pick this commit and try again https://github.com/espressif/esp-idf/commit/68fbe1dd83463c9cc636fcfab3075f6380306d0e

aaa147147, 2025-02-24:
Thanks, it is good

closed 2025-02-24 by dhrishi