espressif/esp-matter #1287: Acessory not registering with home hub

jonhkr, 2025-02-21:
I'm implementing a window covering device using an esp32c3 board.
It works when I pair it with homekit with my phone, but it is not registering with the home hub. It only works on my phone.

I keep seeing these errors in the logs:

```
I (1256040) chip[EM]: >>> [E:54685r S:0 M:12518899] (U) Msg RX from 0:60A76FCBFB254516 [0000] to 0000000000000000 --- Type 0000:30 (SecureChannel:CASE_Sigma1) (B:196)
I (1256040) chip[IN]: CASE Server received Sigma1 message . Starting handshake. EC 0x3fc9eee0
I (1256050) chip[EM]: <<< [E:54685r S:0 M:247621192 (Ack:12518899)] (U) Msg TX from 0000000000000000 to 0:60A76FCBFB254516 [0000] [UDP:[FE80::8EC:8D38:6D91:C5E7%st1]:52884] --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
I (1256070) chip[SC]: Received Sigma1 msg
E (1256080) chip[SC]: CASE failed to match destination ID with local fabrics
I (1256090) chip[EM]: <<< [E:54685r S:0 M:247621193 (Ack:12518899)] (U) Msg TX from 0000000000000000 to 0:60A76FCBFB254516 [0000] [UDP:[FE80::8EC:8D38:6D91:C5E7%st1]:52884] --- Type 0000:40 (SecureChannel:StatusReport) (B:34)
I (1256110) chip[EM]: ??1 [E:54685r S:0 M:247621193] (U) Msg Retransmission to 0:0000000000000000 in 407ms [State:Active II:500 AI:300 AT:4000]
E (1256120) chip[IN]: CASE Session establishment failed: 10
E (1256130) chip[EM]: OnMessageReceived failed, err = 10
I (1256140) chip[EM]: >>> [E:54685r S:0 M:12518900 (Ack:247621193)] (U) Msg RX from 0:60A76FCBFB254516 [0000] to 0000000000000000 --- Type 0000:10 (SecureChannel:StandaloneAck) (B:26)
```

Can you help me understand what's going on?

Jerry-ESP, 2025-02-25:
Can you describe the procedure in detail? The log means that the device receive a message with a node id different with the local one.

dhrishi, 2025-03-11:
@jonhkr Please provide the requested details or close the issue if it works

closed 2025-03-21 by jonhkr